malnvenshorn/OctoPrint-FilamentManager #1: Currency symbol is Euro

ctgreybeard, 2017-08-22:
Just a cosmetic thing but we use $ here in the US. Could you use the locale currency symbol?

malnvenshorn, 2017-08-23:
The issue is I don't know if there is a way to get the currency symbol for the browser language and I don't want to create a list which maps the language codes to the currency symbols, because in Europe alone there exists more than 20 different currencies. Of cause I could add an input field to make the symbol configurable, but I'm not enthusiastic about that idea either. I think it doesn't matter what currency symbol is shown, because it's just an input field in the settings. 

I'm still open for suggestions. Maybe I'm adding a universal symbol for money.

Mavyre, 2017-08-23:
The best option is to let the user input its own Character as the currency. It'd be the same as adding the universal symbol.

ctgreybeard, 2017-08-23:
I'll take a look at it. This clearly isn't a critical issue so it doesn't need immediate attention. If I find an answer I'll post back.

peepsnet, 2017-08-23:
where are the files on the RPI(OctoPI) I can just edit it for now??

ctgreybeard, 2017-08-24:
The character is in the file templates/filamentmanager_spooldialog.jinja2 around line 42.

If you change it manually you will need to re-change it after every release.

peepsnet, 2017-08-24:
Where in the file structure of the RPI is that file?? I cannot find it

peepsnet, 2017-08-24:
I found it burried in
`/home/pi/oprint/lib/python2.7/site-packages/`

cbale2000, 2017-08-26:
Other cost calculator plug-ins just have an open field where you can enter any symbol (though many have it set to Euro by default). I'll second this feature request.

malnvenshorn, 2017-08-26:
Currency symbol can be customized in version 0.2.0

closed 2017-08-26 by malnvenshorn