malnvenshorn/OctoPrint-FilamentManager #3: Multiple printers?

KD7LRJ, 2017-08-23:
Could the database be hosted on a single server and accessed by multiple OctoPrint installations?

malnvenshorn, 2017-08-23:
I would say yes and no. The plugin uses a SQLite database, which allows an unlimited number of simultaneous readers, but only one writer at precisely one point in time. In theory you could put the database on a network share, but if the network filesystem doesn't lock the file correctly your database can be corrupted. A client/server database system like MySql might be the better solution in this case, but this is not supported at the moment.

papinist, 2017-08-27:
This would be a great feature: sharing the filament database between mutiple printers since our filaments can be used by one or another printer (for those who have multiple printers).
This could be done with DB hosted on a network share (e.g. a nas) or with a server\client structure of your plugin, i.e. printer #1 has the server side, printer #2 and #3 have clients side.
There should not be issues about mutiple writings about the same filament since a filament spool can be used on a single printer at a time.

Great plugin BTW!

malnvenshorn, 2017-09-05:
It takes quite an effort to implement this feature. Rewriting the database class to support e.g. MySQL is the easiest part. More challenging is the question how we keep the data consistent between all clients. I think websockets is the way to go here. This requires to write a separate server application which does all the database work and notifies the clients on changes.

To sum things up, it is not very likely that this feature comes in the near future. It is not particularly high on my to-do-list, because I, personally, don't need this.

malnvenshorn, 2017-11-19:
Just pushed a first (hopefully) usable version of this feature to the repository. Partly it is a complete rewrite. Because of this the version might contain several bug. Therefor I would be pleased if some of you would test it. But be aware that I won't use it in production. You can install it using the this link: https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/feature/PostgresSupport.zip

I decided to support an external DBMS instead of a client/server solution, because with the latter I had to implement and manage a socket between the instances to notify each one when the data is altered. At the moment only PostgreSQL is supported. I picked this DBMS because of it's build-in notification feature. Which saved me some work and notifies almost instantly. Don't ask for other DBMS, e.g. MySQL. They will most likely not come.

malnvenshorn, 2017-11-21:
Merged with `develop` branch

malnvenshorn, 2017-12-07:
I don't intend to provide support how to setup PostgreSQL. You can find tutorials using your favorite search engine. But if someone wants  to write a guide for the wiki feel free to do so.

closed 2017-12-12 by malnvenshorn