malnvenshorn/OctoPrint-FilamentManager #4: Frontend Popup Selection / temp control

PowerShellCrack, 2017-08-23:
I think this plugin is an awesome idea. I currently the print history plugin which allows spool selection, but it doesn't record how much is used and what is left. This plugin would be awesome is I was able to select the spool right before I start the print; instead of going into settings and changing the tool. 

Another added feature to this database would be print temperatures for each spool. Then maybe when  popup message come up prior to finalizing the print queue asking what filament is being used. This could ultimately control the temp settings as well. 

Here is an idea I was thinking of an ran across yours. https://github.com/foosel/OctoPrint/issues/2092. I made an image of what I wanted the popup to look like: [https://goo.gl/photos/fishasnpCTD3Rajg8](url)

malnvenshorn, 2017-08-24:
> This plugin would be awesome is I was able to select the spool right before I start the print; instead of going into settings and changing the tool.

I know that it seem not very convenient that you have to open the settings to change the selected spools. I thought about adding this to the sidebar, but the selection would be persistent. It gets only saved when you press 'save' in the settings window, because I'm using the 'config.yaml' to store those selections. Maybe I should store it in the database to, but I have to think about it a bit more.

> Another added feature to this database would be print temperatures for each spool. Then maybe when popup message come up prior to finalizing the print queue asking what filament is being used. This could ultimately control the temp settings as well.

I'm not going to implement such a feature, because that's what slicers are for. For different material you need not only the temperature. Flow rate, cooling fan settings, retraction rate, etc. might differ also. To sum it up you have to slice your model again and for this you have the Cura or Slic3r plugin.

jasoncross, 2017-08-24:
> I'm not going to implement such a feature, because that's what slicers are for. For different material you need not only the temperature. Flow rate, cooling fan settings, retraction rate, etc. might [...]

I disagree.  For the same model in PLA, for example, I almost never re-slice.  Why would I if the only difference is optimal temperature?

PowerShellCrack, 2017-08-25:
That's what I was saying. I think alot if people don't change their infill retractions, etc that often. Usually oncdbuvgeg it right it's good for most prints. But we are constantly changing filament and temps. I just thought that with the ability to record filament usage, controlling the temp as well would help reduce failed prints. It would be an automated way of helping typical uses.

 Or maybe have a confirmation popup confirming filament spool and print temp. Again this is just a thought. I'm not that good at writing code for the octoprint, but I'm learning. Hopefully when I do, I will write it as my own plugin.

jcjenk, 2017-08-25:
I think having another column for preferred temperature is a great idea.  It doesn't have to make any changes to the print/slice but it would be a reminder to me on the temp that I had success with.  Why not add another column for "notes" where I could write anything I want about the spool (where I bought it, my thoughts on it, anything to remind me.  I have about 6 different spools I'm using and can't keep it all straight.

malnvenshorn, 2017-08-30:
In the next version of the plugin you can set a temperature offset for each spool which gets automatically applied on spool selection.

jasoncross, 2017-08-31:
Excellent. Glad to hear it.

closed 2017-10-18 by malnvenshorn