malnvenshorn/OctoPrint-FilamentManager #6: Offset Hotend and Bed by Filament

jasoncross, 2017-08-24:
Less an issue and more an enhancement request.  

Many filaments are slightly different for optimal temps.  But uploaded data files to print are often uploaded with a specific temp.

For example, I have a model uploaded to Octoprint with the default 190 temp for PLA but have a black PLA that is best at 205 degrees.  Right now I have to manually recall to offset 15 degrees when using that filament.  It would be great if when setting a profile up for a filament, I could set a default offset and have that applied when printing with that filament.

malnvenshorn, 2017-08-24:
Sounds quite the same as #4. The thing is it will (most likely) only work for the same material from the same vendor. For example if you use PLA from a different vendor chances are high that you need to adjust other settings as well, e.g. the flow rate. If you use a different material you definitely have to reslice your model again. So it would be limited in use.

jasoncross, 2017-08-24:
I already put a different profile for each combination of material/vendor, so not really.  For example I have a profile for Inland Silver PLA, Zyltech Transparent PLA, etc.

closed 2017-08-25 by malnvenshorn