malnvenshorn/OctoPrint-FilamentManager #10: Odometer will decrease remaining weight

lchabin, 2017-08-27:
Hi, i want to understand.
I would expect your plugin to decrease the remaining weight of each spool at some point, like at the end of the print. Yet after 5 prints, it seems it does not.  
Am i experiencing a bug here or what ?
It does nothing, expect some weird  rounding errors i have seen on the remaining weight (i enter something like 245.45g and later it gets changed in the low digits)
To be sure : are we supposed to make ourselves the subtraction and update the remaining weight ? Weird.

malnvenshorn, 2017-08-27:
You shouldn't need to update the remaining filament manually. The plugin calculates the used filament from the gcode send to the printer and updates the spool automatically, when the print is paused, finished, or failed.

If you're printing from sd card please see issue #9, otherwise I need some more information, like log file and output of the javascript console.

lchabin, 2017-08-27:
No I am not printing from an SD Card.  

Actually the weight changes, but by a wrong value.
I have done the 3 fw upgrades, with no effect.

I have done again a print this evening, and the weight has changed from 950g
to 949,9248349413936 g ... (note the coma separator)
for a print that weighted actually : 9.85g (computed by sliv3r original prusa i3)

I can do a skype session tomorrow if you can, any time. i am on holidays. To show you my screen.

A unit problem ? A problem with the fact we have in France ',' instead of '.' for the decimal separator
Log file of what ? Octoprint ? your plugin  has a log ? is it kept on reboot ? kept on fw upgrade ? where is it ? for the javascript console, you are asking too much (now)

lchabin, 2017-08-27:
Here is octoprint's log

[octoprint.log.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1254771/octoprint.log.txt)

slayer1551, 2017-08-27:
I've noticed this too, had a new spool with 1kg did a couple of prints and its still at 1kg. I've had to manually update my remaining amount. Printing from octoprint directly not an SD card

ctgreybeard, 2017-08-28:
+1
I thought that because I was running using the virtual printer that was causing this but maybe not. In my testing the weights do not change.

malnvenshorn, 2017-08-28:
> 2017-08-27 20:54:01,468 - octoprint.util.comm - INFO - Finished in 6040.937 s.
2017-08-27 20:54:01,574 - octoprint.plugins.filamentmanager - INFO - Filament used: 12.5 mm (tool0)

Looks like an issue with the gcode parser. If someone could provide a gcode file this would help me to reproduce this problem. I haven't noticed this behavior, but it seems to affect a lot of people.

lchabin, 2017-08-28:
That's the one. All gcode for me were produced by Prusa3D Slic3r MK2. All had the problem I think.
[LACK_filament_guide_top.zip](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1256487/LACK_filament_guide_top.zip)

lchabin, 2017-08-28:
Ohh, pay attention in the same log, that another print that lasted 39827.697 s was also analyzed to use also and only 12.5mm of filament !

commit b6b8b028 (2017-08-28): Fixed #10
closed 2017-08-28 by malnvenshorn