malnvenshorn/OctoPrint-FilamentManager #14: fails with large gcode files

Andrei-Pozolotin, 2017-08-31:
when gcode file size is large (~15MB)
plugin stops reporting length/weight and stops giving warnings

[image]

malnvenshorn, 2017-08-31:
It looks more like OctoPrint hasn't finished parsing the gcode file to generate the metadata. Does OctoPrint shows the info if you disable the plugin? If so you need to provide more information (OctoPrint.log, output of the javascript console).

Andrei-Pozolotin, 2017-09-01:
thank you for looking into this

1) file to reproduce the issue attached: [exp-1.gcode.zip](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1268918/exp-1.gcode.zip)

2) it looks like octoprint finished parsing, since g-code viewer shows layers, while plugin does not show any stats at the same time

3) the stats do show up at the end of the print
[image]

logs are lost, will try again later

malnvenshorn, 2017-09-01:
That the approximated print time is missing strongly indicates that the analysis isn't done yet. This is backed up by the fact that the stats are shown after the print has finished. The analysis has a very low priority, because it needs a lot of cpu power. If you start your print before the analysis is done the generation of the metadata gets delayed (this avoids interrupting the print process). In your case the analysis is finished after your print.

I tried your gcode file and everything works as expected. There for I'm marking this as "not a bug", till I got information which indicate something else.

Andrei-Pozolotin, 2017-09-03:
most slicers add a summary header to g-code:
```
;FLAVOR:RepRap
;TIME:946
;Filament used: 0.633888m
;Layer height: 0.2
;Generated with Cura_SteamEngine 2.6.2
```
you could use a 2-step info display:
show summary based on:
1) slicer header (takes no time)
2) full octopirnt analysis (depends on file size)

malnvenshorn, 2017-09-03:
What you suggest is not feasible. Every slicer is different. They don't generate the same information in the same format at the same place if they provide them at all. Besides that the plugin only uses the stats which OctoPrint provides and I'm not going to change that.

I opened a brainstorming ticket on the OctoPrint tracker to discuss ways to speed up the metadata generation foosel/OctoPrint#2097

closed 2017-09-06 by malnvenshorn