malnvenshorn/OctoPrint-FilamentManager #16: Even though a new spool is selected on sidebar, it decrements previous spool

lchabin, 2017-09-02:
Even though I am logged and have selected a new spool, on every print, it is changing the amount from the previous spool. Selecting again did not solve (at least once). I did 4 prints like that. 
Is there a sequence to respect for the selection to be taken into account in effect ? 
Opening on my mobile phone (not logged in) the url shows the previous spool :  sort of proof that what you see is not what you get.
A log file with the weight subtracted and spool NAMES (not just tool 0)  would help in this phase where there are still annoying bugs. (makes my life harder than with no plug-in)

malnvenshorn, 2017-09-02:
Have you tried the latest version from the develop branch (requires OctoPrint 1.3.5rc)? It contains some changes and fixes.

jasoncross, 2017-09-05:
Ah, I've been running into the same issue.  Not running RC.  It's frustrating because once it's off, it's a bear to correct accurately, which throws off trust in the count from there on out.  Maybe a screen that shows the usage by print that could be used to update the spool odometer manually would help with that.

malnvenshorn, 2017-09-05:
Please try the version from the develop branch. It contains the next release (just waiting for OctoPrint 1.3.5 to be published).

jasoncross, 2017-09-14:
I'm running 1.3.5 rc3 but no other plugin version is showing up as available for update.

malnvenshorn, 2017-09-14:
Because it is not published yet. Use this link https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/release/0.3.0.zip Please delete your browser cache after installing.

brunofporto, 2017-10-16:
Having the same issue. I'll try the latest RC.

An historic table, that allow us to manually correct it (marking some print to another spool) would be nice too. Even if the bug is corrected we may select the wrong spool by mistake :D

malnvenshorn, 2017-10-16:
> An historic table, that allow us to manually correct it (marking some print to another spool) would be nice too. Even if the bug is corrected we may select the wrong spool by mistake :D

You can weigh the affected spools and manually adjust the value.

reggaemanu, 2017-10-16:
Same issue here, it keep re-selecting an old spool, I'll try the new version...
BTW, I agree with brunofporto, an history would be really great (something like the print history plugin but using your spool managment :) )
Anyway, thanks for this awesome plugin, it's really useful !!!

laurent-chabin-sb, 2017-10-16:
I have installed the RC since sept 5th. No problem anymore. Except one maybe in ~20 prints.
Now like the others I guess, the remaining problem is that in the many prints i have done, after a filament change, I have forgotten to change the selection before starting the print. It happens to me almost 50% of the time. Why ? Because I am a human. I want to print asap because the filament is cooking.  10 minutes into the print, I always realize my mistake, but then, it is too late. The menu is grayed. 
+1 if you have encountered that problem many times.
In any case, A history log file is a must because I think there are remaining issues. 
There is also the case of when I change the filament during the print for multicolor. I can understand that you don't want to handle the complexity of that.

malnvenshorn, 2017-10-16:
In addition to my previous post. The amount of filament used can also be found in the log file of OctoPrint and can be used to correct the mistake. I have to admit that the logging can be improved and I'm setting this on my to-do-list.

Despite that I'm open for any suggestion which would help to prevent having the wrong filament selected in the first place.

@laurent-chabin-sb 
Multi color prints are not an issue. If you pause the print the amount of used filament get's updated to the currently selected spool. So just change the filament and select the new spool after that.

@reggaemanu
I understand that this might be great, but I like to keep things simple. This plugin is not meant to provide some kind of print history.

lchabin, 2017-10-16:
Ahh. I had not noticed that when you pause the print, the spool selection become selectable. I will do that now, when i forgot. I will check what you say is correct. Seems to be for the first pool. True for the next one ? I will check and report if i see a problem.

malnvenshorn, 2017-10-28:
I would like the get some more feedback if the issue persists in the current plugin version or not, otherwise I tend to close this issue.

lchabin, 2017-10-29:
I made several prints, but unfortunately, did not check for consistency. Can't help.

lchabin, 2017-11-05:
I am still one step late here in my general knowledge of octoprint. How can I install the dev version of a plugin, Do you have the url of a zip file ?

malnvenshorn, 2017-11-05:
You can install the latest develop version using this url: https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/develop.zip

dkingsjr, 2017-11-12:
I am having a similar issue. Before initiating the print, I select the spool I want to use, but then the spool selection reverts back to the previous spool "used". (i.e. I select black, but when Press the start print button, it shows green spool instead.

malnvenshorn, 2017-11-12:
What version are you using, 0.3.0 or develop? If the latter, please provide 
a link to the logfile and javascript console output. Otherwise try the 
latest version from the develop branch.

dkingsjr, 2017-11-13:
@malnvenshorn , I am unsure. I know I downloaded it using the plugin manager. my experience with anything linux is very basic, so i am unsure of how to retrieve said information to you. however, only about an hour ago (as of the time of this comment), i saw you pushed an update. so far it seems to have remedied the issue. I will let you know my findings soon. thanks, Dave

malnvenshorn, 2017-11-13:
@dkingsjr You have to manually install the develop version using the link I provided a few comments earlier. But if you have installed the latest update (0.4.0) there is no need for this. I appreciate your feedback. Maybe this ticket can finally be closed.

dkingsjr, 2017-11-14:
@malnvenshorn , Your update pushed last night seems to have worked! by the way, thanks for producing such a good plugin!

malnvenshorn, 2017-11-23:
Because there are no negative reports since the last commit I'm considering this fixed. Closing.

closed 2017-11-23 by malnvenshorn