malnvenshorn/OctoPrint-FilamentManager #17: keeps losing everything

scottjl, 2017-09-05:
i've set up this plugin twice, and twice it has lost all my added spool information. everything was input, after job finished, everything was gone. i'd be happy to help diagnose the issue, i can't use the plugin if it keeps losing my data.

malnvenshorn, 2017-09-05:
If you manually delete the database you also have to remove the "_db_version" entry in the config.yaml (will be fixed in the next release). But why did you removed the database in the first place? Could you explain "everything was gone" a bit more? What did you do (step by step)? And please provide the octoprint.log too.

scottjl, 2017-09-06:
Ok. Better explanation. I had first installed the plugin, set up about a dozen spools with 3 manufacturers in. Worked fine for a few prints. Then suddenly my spools were gone. Selection drop-down on main page was empty. Went into plugin configuration, all data was gone. Couldn't add a new spool either, "unexpected database error". Rebooted. Same situation. Uninstalled and installed plugin. Same situation. Went poking around, found the filament.db file, removed it, reinstalled plugin, rebooted. All was well.

Added a few spools the second time around. Again. Had a few prints and all was fine. Then again, spools disappeared, no longer in settings, can't add again. This time I simply removed (actually simply renamed) filament.db and restarted octoprint and the plugin is working again, but I'm reluctant to use it if I'm just going to keep losing my data every few prints.

I can provide the bad database file and logs if you wish.

malnvenshorn, 2017-09-06:
Sound really strange to me. Please provide the log file and the output of the javascript console after the issue appeared. The database could help too. Currently I have not the slightest idea what could be the cause of this.

dkingsjr, 2017-09-15:
@malnvenshorn , I have the same problem as well. I added the filament profiles for 2 types of filament. then added 7 spools, and they showed up briefly, and now they will not show up at all...I attempted to reinstall Filament Manager, but the issue is recurring. I would post pictures of what i see, but I don't know how to.

laurent-chabin-sb, 2017-10-16:
Hello malnvenshorn, this is scary.
What can we do to save the spools data/profiles and recover it in case of any type of corruption. 
Is it possible to do that simply, Just copy some files ?

malnvenshorn, 2017-10-16:
You can simply backup the whole database. It's located under _\<basedir\>/data/filamentmanager/filament.db_. For the location of the basedir see [OctoPrint docs](http://docs.octoprint.org/en/master/modules/settings.html#octoprint.settings.settings).

malnvenshorn, 2017-10-27:
I'm closing this now, because no further information were provided for over a month. If someone provides new information I will reopen this ticket.

closed 2017-10-27 by malnvenshorn