malnvenshorn/OctoPrint-FilamentManager #18: No Filamentcount when Printer request Pause

Coldfinger, 2017-09-06:
When the printerfirmware requests a pause during the print, the value for the used filament is not correctly substracted from spool, after resume and finish the print

malnvenshorn, 2017-09-06:
Please provide some more information. Describe you situation, what did you do, what did you expect, what happened instead? Please also post your octoprint.log

Coldfinger, 2017-09-06:
[octoprint.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1282210/octoprint.txt)
THe situation is, that the weight of the spools dont change after successfull prints, on my Ultimakers, but on my FLSun Cube. After a few tests with other addons and fresh installed octoprint server, i found out, the modified Firmware of my Ultimaker send,  short after start of the print a pause-command (//action:pause) to octoprint, a few seconds later a resume-command. In this case the Filament Manager Plugin dont substract the used amount if filament from selected spool, after finishing the print.

malnvenshorn, 2017-09-06:
This will not be easy to figure out. It's looks firmware specific and because I don't have an Ultimaker I cannot reproduce this behavior. Could you run OctoPrint with the "--debug" flag and post the logfile again?

Coldfinger, 2017-09-06:
I think the same behavour will apear, wenn the print will paused from octoprint, and then resumed, but im not sure. I test a couple of things tomorrow, to give you additional informations.

malnvenshorn, 2017-09-06:
Pausing/Resuming in OctoPrint works. At least I never had an issue ;-)

Coldfinger, 2017-09-07:
[image]

OK,, i have tested some things. The Problem seems to be the resume command.

My Slicer (Simplify3d) sets a G4 Command at the beginn of GCode-File. The Printer tells on G4 the Host to pause the Print, and resume it after 500ms. This Action Commands are implemented in Marlin and Octoprint.

I have modified the Firmware on one of my printers, so it dont send Action commands on G4 to Octoprint and now the Filament Manager is counting the Filament correctly. I have tried to remove only the Pause Command but then it dont count the used Filament on Spool

Perhaps it is reproducable on Printers with Marlin-FIrmware >v1.1.4 when send a M118 command with action:resume

I hope this Information will help a bit. For me the Workaround is not perfekt, but it works, my Printer send only Pause-Commands when filament runs out.

malnvenshorn, 2017-09-08:
I think I found the cause of the problem, but I don't know if I can do something about that.

```
2017-09-08 11:39:11,017 - octoprint.plugins.filamentmanager - DEBUG - PrintStarted
2017-09-08 11:39:11,019 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-08 11:39:46,119 - octoprint.plugins.filamentmanager - DEBUG - PrintResumed
2017-09-08 11:39:46,231 - octoprint.plugins.filamentmanager - INFO - Filament used: 9.49335 mm (tool0)
2017-09-08 11:39:46,235 - octoprint.plugins.filamentmanager - DEBUG - PrintPaused
```

I'm trying to explain this behavior. As you can see the _PrintResumed_ event is fired before the _PrintPaused_ event. This happens because of the combination of three things. First the odometer gets disabled on pause, because I don't want to track the filament on possible spool changes (this would distort the filament value). Second OctoPrint sends a M114 command on pause to get the current position. The event is fired after that. Third the time of 500 ms between _action:pause_ and _action:resume_ is to low. Therefore the answer of the M114 command comes after the print is already resumed.

I testet this using OctoPrints virtual printer with:
```
!!DEBUG: action_pause
G4 P500
!!DEBUG: action_resume
```

malnvenshorn, 2017-09-08:
If you are resourceful you could try the develop branch and test if it fixes your problem. But it requires OctoPrint 1.3.5rc and you might need to clear your browser cache.

Coldfinger, 2017-09-08:
After Update Octoprint too 1.3.5RC and install Development-branch of Filament-Manager, i become an error on creating a new spool: SQL Error: table spools has no column named temp_offset
2017-09-08 14:40:03,816 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/spools

Uninstall, and then install new dont change this, can I manual delete the Database, and then install Filament Manager new?

malnvenshorn, 2017-09-08:
Please try to set the following in your config.yaml

```
plugins:
  filamentmanager:
    _db_version: 1
```

Coldfinger, 2017-09-08:
Okay, the development-version works fine.

Thanks for the perfect support.

malnvenshorn, 2017-09-08:
You're welcome

closed 2017-10-18 by malnvenshorn