malnvenshorn/OctoPrint-FilamentManager #22: Filament List is empty after upgrade to v0.3.0

stelgenhof, 2017-10-17:
After upgrading OctoPrint to version 1.3.5 and Filament Manager to 0.3.0, my list of spools are not visible and OctoPrint just shows a continuous spinner with the dropdown list being empty. Also the list of spools via the Settings menu is empty (See screenshot).

[image]

malnvenshorn, 2017-10-17:
Please provide all information as requested in the issue template.

stelgenhof, 2017-10-17:
Please disregard this issue. After I cleared my browser cache and refreshed the page, the list of filaments are visible. (All is working well)

malnvenshorn, 2017-10-17:
I reopened this, because maybe people will read this before opening another issue.

apazga, 2017-10-17:
Thanks @malnvenshorn & @stelgenhof for the solution. I'll try it this afternoon.
Keep up the good work! ^_^

malnvenshorn, 2017-10-30:
Added FAQ entry in the readme

closed 2017-10-30 by malnvenshorn