malnvenshorn/OctoPrint-FilamentManager #25: Icon in sidebar is broken

larp-welt, 2017-10-18:
#### Error

The icon in the sidebar is missing.

[image]

#### Reason for the missing icon

The rease for the missing icon is: The plugin tries to use a icon "fa-reel".

`<i class="fa icon-black fa-reel"></i> Filament Manager`

But there is no icon "reel" in the Fontawesome font.

The icon is defined inside __init__.py line 108:

`dict(type="sidebar", icon="reel", template="sidebar.jinja2", template_header="sidebar_header.jinja2")`

malnvenshorn, 2017-10-18:
The icon is defined in _static/css/font.css_ and is shown fine in my browser. Please check if the style sheet and the font file is loaded correctly. What browser do you use?

larp-welt, 2017-10-18:
Ahhh, that's interessting!

Octoprint doesn't load any file `static/css/fonts.css` on my system, but a file `http://servername.local/static/webassets/packed_plugins.css?4c88f514`. That file contains the lines:

```css
@font-face {
  font-family: 'icomoon';
  src:  url('../../plugin/filamentmanager/static/fonts/icomoon.eot?2tngg');
  src:  url('../../plugin/filamentmanager/static/fonts/icomoon.eot?2tngg#iefix') format('embedded-opentype'),
    url('../../plugin/filamentmanager/static/fonts/icomoon.ttf?2tngg') format('truetype'),
    url('../../plugin/filamentmanager/static/fonts/icomoon.woff?2tngg') format('woff'),
    url('../../plugin/filamentmanager/static/fonts/icomoon.svg?2tngg#icomoon') format('svg');
  font-weight: normal;
  font-style: normal;
}

.fa-reel {
  font-family: icomoon;
}

.fa-reel:before {
  content: "\e900";
}
```

But using the web developer tools, I can't see any access to the one of the font files. :-(

I'm using Chrome 62 under Ubuntu 17.04.

larp-welt, 2017-10-18:
Argh, sorry, silly me! It loads the font file just fine! 

But it doesn't show the icon!

larp-welt, 2017-10-18:
That is, what is looks like in Firefox under Linux:

[image]

That is Edge under Win10:

[image]

And that's IE 11 under Windows 10:

[image]

malnvenshorn, 2017-10-19:
Just tested it with Chrome, Firefox and Vivaldi, but i can't reproduce this. The icon loads correct.

larp-welt, 2017-10-29:
Strange! I've setup my octoprint server completly new, and now I can see the icon too! 

Must have be a strange error in my setup!

closed 2017-10-29 by larp-welt