malnvenshorn/OctoPrint-FilamentManager #26: Cancelled print still subtracts print requirements from roll

beercity, 2017-10-19:
I cancelled a job right after starting it.  I was using a brand new spool, and filament manager subtracted the requirements for the job from the spool.

I suspect there should be a way to subtract only the used filament from the spool. If it is not possible to do this, at least an estimate based on % of job completed.

I am using the latest Octoprint as well as Filament Manager versions.

malnvenshorn, 2017-10-19:
It only counts the used filament so this shouldn't be an issue. Please stick to the issue template and provide all information, otherwise I can't help.

beercity, 2017-10-19:
I have a long print running now but I will get a good test of the scenario once I am able.

malnvenshorn, 2017-11-02:
Closed due to the lack of information. I will reopen the ticket if you provide ALL required information.

closed 2017-11-02 by malnvenshorn