malnvenshorn/OctoPrint-FilamentManager #28: No count of filament with prusa i3 mk2 multicolor

patrickfanz, 2017-10-25:
I have Octoprint version: OctoPrint 1.3.5 (master branch)
Filament Manager: 0.3.0
Printer: prusa i3 mk2 multicolor (4 colors)

Filament-Manger does not count the selected Filament on the main page at, where is written "Tool 0".
After a print is always written the same value of gram.
In the settings I have added all my filaments, but I have to correct the values of rest of the filaments by hand.
[image]

malnvenshorn, 2017-10-25:
Please stick to the issue template and provide all information, otherwise I can't help.

malnvenshorn, 2017-11-08:
Closed due to the lack of information. I will reopen the ticket if you provide ALL required information.

closed 2017-11-08 by malnvenshorn