malnvenshorn/OctoPrint-FilamentManager #29: Plugin name for config.yaml - reorder in sidebar

pandel, 2017-10-25:
Hi Sven!

First, many thanks for your great plugin. Absolutely useful addition to OctoPrint!

One question: with the last update you changed the fixed position of the plugin to a configurable one. I tried to reorder the plugins in the sidebar, but somehow it did not work. I am not sure what I did wrong....

Could you be so nice to tell me the correct plugin name to use in the appearance section? Or do you have an example on how to set this up for the old plugin position? Sorry for the question, but I am not used to modify config.yaml directly...

Many thanks,
Holger

malnvenshorn, 2017-10-25:
Of course. To restore the old position under the printer state you need the following in your config.yaml

```
appearance:
  components:
    order:
      sidebar:
      - connection
      - state
      - plugin_filamentmanager
```

pandel, 2017-10-25:
Arggg! Many many thanks for you ultra fast answer! I did not know that I had to prepend the plugin name with „plugin_“! That was my fault!

closed 2017-10-25 by pandel