malnvenshorn/OctoPrint-FilamentManager #30: Support for Multi-Material (Prusa i3MK2S MM)

mfmjos, 2017-10-30:
This is really a feature request. I didn't see a proper place of that type of thing.

#### What were you doing?
Printing objects using multiple filaments

#### What did you expect to happen?
I would like to track multiple filaments in the same print.

#### What happened instead?
Currently it appears that FilamentManager only tracks one spool at a time.

malnvenshorn, 2017-10-30:
The odometer feature supports multiple extruder. If you encounter issues 
please file in a complete bug report.

mfmjos, 2017-10-31:
Thank you very much for responding. I didn't have my extruder count set correctly (I just upgraded to 4 materials). Now I see that filament manager does exactly what I need.

closed 2017-10-31 by malnvenshorn