malnvenshorn/OctoPrint-FilamentManager #33: No filament amount reduction, OctoPrint 1.3.5 (master branch)

GitPetri, 2017-11-05:
Before opening a new ticket please take a look if there is already one addressing the same concern.

If you want to report a **bug** stick to the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the FAQ and release notes.

If you have an other request mark them with one of the following prefixes in the title:

- "[Request]" for a **new feature** request
- "[Improvement]" for suggestions for an **existing feature**
- "[Question]" for a **question**

---

#### What were you doing?
Print any 3D model

#### What did you expect to happen?
Filament manager reduces amount of filament on selected spool based on usage

#### What happened instead?
Filament manager doesn't reduce amount of filament for the selected spool

#### Link to octoprint.log
[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1444160/octoprint.log)

#### Link to gcode file
n/a

#### Link to contents of Javascript console in the browser

n/a

#### Further information

The problem started happening after OctoPrint 1.3.5 update, earlier it worked ok.

malnvenshorn, 2017-11-05:
The log doesn't show any errors. It looks like there is either no spool selected or the odometer feature is disabled.

GitPetri, 2017-11-05:
@malnvenshorn : Thanks, for some reason the odometer had been de-selected during one of the updates. Issue can be closed as a SUE (stupid user error)

closed 2017-11-05 by malnvenshorn