malnvenshorn/OctoPrint-FilamentManager #35: TouchUI compatibility

sticilface, 2017-11-10:
Currently it does not appear as an option in TouchUI.  Other plugins do. 

could it be possible to extend filament manager to give a menu option in touchUI? 

Tx

malnvenshorn, 2017-11-10:
I don't think this has anything to do with my plugin. You should ask the "TouchUI" developer to integrate this.

sticilface, 2017-11-10:
I will do! Fair point

jneilliii, 2018-07-21:
@malnvenshorn, just in case you wanted to accomplish this yourself, I just figured out how to do it for my plugin's sidebar panel.  I added the following to my view models js file.

```
		self.onStartupComplete = function () {
			// move elements if using touchui
			if($('html[id="touch"]').length > 0){
				$('#touchui_dropdown_link').before('<li id="sidebar_plugin_bedlevelingwizard_wrapper_link" class=""><a href="#sidebar_plugin_bedlevelingwizard" data-toggle="tab"></a></li>');
				$('#tabs_content').append($('#sidebar_plugin_bedlevelingwizard').addClass('tab-pane'));
			}
		}
```

You would just need to replace the bedlevelingwizard id with your filament manager's sidebar id.

malnvenshorn, 2018-07-23:
Thanks. I'll look into it once I'm back from vacation. Reopened for now.