malnvenshorn/OctoPrint-FilamentManager #36: [Request] Remedy for those who permanently forgot to change the filament

malnvenshorn, 2017-11-11:
I'm opening this ticket to collect ideas and opinions to improve the situation on this matter. Currently I'm tending to the following changes:

- The selected spools will not be saved permanently anymore (if you restart OctoPrint no spool will be selected)
- Add an option to reset the selected spools after the print stops/finishes (or get paused?!) 
- Show a dialog where you have to confirm a print start/resume if no spool is selected

I would be pleased to get some feedback on this. I'm also still open minded for other ideas you might have.

sticilface, 2017-11-12:
This would be great.  my thoughts are as follows ( I always forgot to change it, mainly because it was listed at the bottom under the file manager - which i have now changed.) 

personally I would like it to pop up when you print and say..  continue with this spool or select from this drop down, so you have to either click OK or change the spool. 

Other options could maybe keep a list of prints with times and associated spools.  So at least if you forget to change it you can go back and change it for a previous print, or mid print.  obviously the warnings won't be valid but at least it will prevent you becoming really out of sync.

malnvenshorn, 2017-11-12:
First of all I appreciate your feedback :thumbsup:

> personally I would like it to pop up when you print and say.. continue with this spool or select from this drop down, so you have to either click OK or change the spool.

Two thoughts about this. 

First I'm going to keep the selection in the sidebar, simply because of my cost estimation plugin so you can see the cost before starting the print. If I add a selection popup then we have two places for the same setting and I don't think this is a good interface design. 

Secondly if such popup appears every time I'm afraid that people will simply click "continue" without reading the popup. Maybe because there are annoyed of this or they want to start the print as fast as possible. Resetting the selection will force them to select a spool every time they start a print. It could also be an idea to prevent printing at all if no spool is selected.

> Other options could maybe keep a list of prints with times and associated spools. So at least if you forget to change it you can go back and change it for a previous print, or mid print. obviously the warnings won't be valid but at least it will prevent you becoming really out of sync.

This idea came up in another ticket too (#16). My answer to that was that I'm not going to provide some kind of history. I can almost guarantee that the next request will be to add some kind of graphic evaluation. Suddenly we have a fully fletched history plugin integrated and this is something I don't want.

There might be also some cases where such a history wouldn't help either, e.g. when you have multiple spools of the same color.

TorranceTM, 2017-11-13:
I forget to change the selected filament approximately 10% of the time - No big issue I weigh the spools occasionally when swapping and update the database.

I 100% remember these 10% issues when I hit the print button and the print then starts :-(

I agree that having a popup when print is hit is not good UI/UX, but being human it is always good to have a reminder. So maybe having a setting of "Would you like to be reminded of filament selection when print is started in the settings" would get around this issue for those that would not like to see the popup?

Personally I would like the reminder and a selection box of "choose filament (drop down) or the one currently selected is correct"

Just my 5c - smallest coin available in Australia :-)

malnvenshorn, 2017-11-13:
@TorranceTM So you are also not preferring the "reset method". Did I understand you correctly?

TorranceTM, 2017-11-13:
@malnvenshorn - Correct I don't think I would like the "reset method" because if I am using the same spool (which is often the case) then I don't have to hunt for it in the spool list each time.

My preference would be for the popup reminder selection option.

felixleong, 2017-11-15:
Personally would not like the reset option either because I also tend to leave the spool as is unless I have to change it. I don't mind being an option in the settings to opt-in for that behaviour, though, as I can see certain workflows would appreciate such a option.

Would agree with @malnvenshorn that a pop-up isn't ideal too as I do find myself skipping these dialogs hastily.

With my current workflow w/ the printer, I think I would probably like to see it implemented in the following fashion:
1. **If I notice that I forgot to switch spool setting mid-print:** That I'm allowed to switch spools from the Filament Manager dropdown after I started a print. In this instance, I don't mind a popup confirming that I've switched a spool to confirm the switch.
2. **If I notice that I forgot to switch spools after the print is completed:** This I'm not 100% sure how I handle it. My best idea would be to have some sort of a button under the dropdown that reads "Correct the spool used for previous print" and it'll probably do the following:

   a. Shows a confirmation dialog - showing the old spool and print weight and new spool and adjusted print weight
   b. Add the print weight back to the old spool
   c. Calculate the new print weight and deduct that from the correct spool

Don't think I need like a whole print history, but caching the last print information would be good enough.

malnvenshorn, 2017-11-20:
@felixleong you are already able to manually correct the remaining filament, because the extruded amount will be logged (more detailed if you enable debug logging).  I think this should be enough.

felixleong, 2017-11-21:
Didn't know that it is logged, but I have already done so using the info from the State sidebar :).

Would still love to see a "Correct the spool used for previous print" button to automate that, or some easier way to do inline edits from the Settings to speed up the process :).

techpad-marcb, 2017-11-29:
I too just realized that i constantly forget to change the spool and notice mid print. I'd prefer the popup to be configurable for those that want to use the feature and those that do not. I don't believe its bad ui at all to have a reminder to select the correct spool. Starting a print is a busy task and easy to forget stuff after fussing around with first layer issues or plugged nozzles. Then to start print and go aww man i forgot to select filament.

Just my 5c lol.

malnvenshorn, 2017-11-29:
I decided to not go for a selection popup because of the following reasons:
- Two places for same settings is bad (sidebar and popup)
- If the spool from the sidebar is preselected in the popup you run into the same problems (simply clicking continue to start a print as soon as possible)
- If no spool is selected per default it ends up being the same as I described in my initial comment

Changing the selected filament during a print is also not an option I'm considering because this might cause some issues. If a new spool is selected there some background calculations and database queries and I don't know how this affects running prints. Also there is always the option to pause and resume the print.

TorranceTM, 2017-11-29:
@malnvenshorn Would it be possible to have a selection in the settings to allow for the popup for those that would like it (as a reminder)?

malnvenshorn, 2017-12-09:
A short update. I rethought this again and I'm going to implement this feature in the following way. There will be a confirmation dialog where you have to select the spools again. The selection must match the selected spools in the sidebar. The confirmation dialog must be enabled in the settings (disabled by default).

malnvenshorn, 2017-12-09:
Just pushed this feature to `feature/confirm-spool-selection`. As always, feedback is appreciated ;-)

sticilface, 2017-12-11:
This may seem like a stupid question.  but how do i go about getting this installed over my existing version on octoprint?

malnvenshorn, 2017-12-11:
Select the mentioned branch then you can find the link by clicking on 'Clone or download' ;-)

closed 2017-12-12 by malnvenshorn

TylonHH, 2018-01-12:
Seems it doesnt work with two hotends?!?!
I still have to double check this, but my printer is busy right now.

malnvenshorn, 2018-01-12:
It should work with multiple hotends, but the dialog will only show the tools which are actually used for the print job. E.g. if you load a gcode file which requires only one tool the dialog will show only this one even if you have more extruders.