malnvenshorn/OctoPrint-FilamentManager #37: Not working in Octoprint 1.3.5 Master Branch

wabbitguy, 2017-11-12:
Prior to updating Octoprint to 1.3.5 the Filament Manager was working perfectly. After updating all the spools I previously entered were gone.

I reinstalled the update to Filament Manager and the spools all came back.

For the prior version of Octoprint, the filament manager was under the State. Now the filament manager was under the Files section. No big deal.

But in spite of the fact I can select a spool, no cost shows in the State area, nor does the spool ever decrease in weight.

wabbitguy, 2017-11-12:
Deleted the plug in, reinstall it. Included the logs. Still dead as a doornail.

Should also be a method to change the denomination to whats relevant for the user.

However with two printers now, the plug in is less useful because I can't find any way of sharing information between the two octoprints. Drat.

[plugin_pluginmanager_console.log.html.zip](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1464994/plugin_pluginmanager_console.log.html.zip)
[octoprint.log-2.html.zip](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1464996/octoprint.log-2.html.zip)

malnvenshorn, 2017-11-12:
> Prior to updating Octoprint to 1.3.5 the Filament Manager was working perfectly. After updating all the spools I previously entered were gone.

> For the prior version of Octoprint, the filament manager was under the State. Now the filament manager was under the Files section. No big deal.

Have you read the release notes and faq? If yes also provide the output of the javascript console.

> However with two printers now, the plug in is less useful because I can't find any way of sharing information between the two octoprints. 

There is already a request for this (#3).

wabbitguy, 2017-11-12:
Read the release notes. Read the FAQ. Can't find any Javascript console in Safari. All Mud. Clear as.

I give up. Closing. Deleting.

closed 2017-11-12 by wabbitguy