malnvenshorn/OctoPrint-FilamentManager #38: Import/Export feature

kfonda, 2017-11-14:
#### What were you doing?

I have 2 issues: 

First, trying to export my inventory failed. I tracked the error to the fact that I included the print temperature in the spool name (e.g. Black - 190°C). It was the degree symbol that caused the failure. While I'm at it, could you add a field for the temperature?

Second, After editing the spools.csv file and putting it back in the .zip file, importing it does not appear to work. The edited items do not show up in the Filament Manager.

#### What did you expect to happen?

Expected edited data to show up in Filament Manager

#### What happened instead?

nothing

#### Link to octoprint.log

https://www.dropbox.com/s/aln0x6mvmfcxdo5/kfonda_11_13_17_octoprint.log?dl=0

#### Link to gcode file

#### Link to contents of Javascript console in the browser

#### Further information

malnvenshorn, 2017-11-14:
Thanks for your report. Will be fixed in the next release.

> While I'm at it, could you add a field for the temperature?

I guess you're referring to the spool list?! Because of the limited space I can not just add another column. But it might be possible to add some notes below each row.

> Second, After editing the spools.csv file and putting it back in the .zip file, importing it does not appear to work. The edited items do not show up in the Filament Manager.

Currently the import doesn't replace existing entries, only insert new ones.

Feel free to submit a separate feature request for each one.

kfonda, 2017-11-14:
I installed the development branch and tested the fix by: 
exporting my database. 
deleting all of my spools.
editing the spools.csv file and adding the °degree symbol back and putting the changes back into the zip file.
importing the filament.zip file.
and everything worked fine. 

This is what I have now...
[image]

closed 2017-11-14 by malnvenshorn