malnvenshorn/OctoPrint-FilamentManager #43: Will not save filament used to the spool info

HarrysWorkshop, 2017-11-18:
Before opening a new ticket please take a look if there is already an existing one addressing the same concern.

Also read the FAQ: https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki#faq

If you have a **request** or **question** mark them with a "[Request]" respectively "[Question]" prefix in the title.

For **bug** reports use the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the release notes.

(Before submitting your ticket, please delete this text up to and including the line too - it's only here for your information)

---

#### What were you doing?
I finally got the plug in to work, not sure what the problem was, but the left screen does show the filament tool (0) and the amount and weight of the filament to be used.  When the job completes, the amount of filament is subtracted from the original amount as shown on the bottom left of the screen.  
However, at the end of printing, I have an ending G code command that moves the Y axis to the end of the stroke and activates a limit switch that shuts down my complete system.  The system seems to shut down before the new value (as shown in on the bottom left of the screen) is copied to the spool table or is saved to the spool info.  Thus on a new print, the spool table shows the original value and not the revised calculated value.  

[Be as specific as possible. Providing a step by step guide will help me to reproduce the issue. This is needed in order to fix the problem. I cannot help you if I don't know what exactly you did to trigger the issue.]

#### What did you expect to happen?

I would like the system to update and save the filament used either during the print or before the last G code is issued so that when my system shut down, the new spool length would be correct.  If there is a work around to this, please let me know.  I have tried putting the ending G code (G90, Y220 'moves Y axis to end to active the limit swich') in both Curi or Octoprint, and neither allows the filament remaining to be copied or saved to the spool information.

#### What happened instead?
Filament remaining as shown on the main page was not copied to the spool info sheet.  Or, it might have been copied but not automatically saved. (actually, I think that might be the case, it is copied by not automatically saved in time.)

#### Link to octoprint.log
I do not have this since the log is deleted when the unit is shut down.  
[Always include the log file.]

#### Link to gcode file

[If you encounter issues with the odometer feature provide a small gcode file which produces the problem.]

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

[If you have more information which could help me to understand/fix the issue feel free to add them here.]

I have read the FAQ and provided all required information.

malnvenshorn, 2017-11-18:
If I understand you correctly your OctoPrint instance is getting stopped when the print finishes. If this is the case I'm afraid that there is nothing I can do. Constantly updating the database is no option do to performance reasons. I have to wait till OctoPrint sends the corresponding event. After that I can update the database.

Maybe something pops into my mind if you describe how you do the automated shutdown.

malnvenshorn, 2017-11-20:
Sorry it seems that I overlooked that you are using a hardware switch to trigger the shutdown As I sad before there is nothing I can do. You might consider to only shutdown your printer and keep the OctoPrint instance running or you could put a delay between the trigger and the shutdown. Closing.

closed 2017-11-20 by malnvenshorn