malnvenshorn/OctoPrint-FilamentManager #45: [Question] How to get it started?

Knusperhas, 2017-11-25:
Hello, 
First: It looks great.

My Odometer is deactivated in the Settings (because i don't have one yet). I added some Spool Data and so on. that works fine. On the Main Page i can choose one of my filaments and he knows it even after a restart.

But if a print is finished he doesn't count down the remaining filament... I've tried different Print jobs and all finished verry well. And the expected filament length is also displayed before i start the Print.

So my question is: Isn't it working without an Odometer?
And for my further steps: Where should i buy an Odometer? and how should it be connected to the Rasperry?

Hope you can help me :)

[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1503267/octoprint.log)

malnvenshorn, 2017-11-25:
You don't need additional hardware, but you must enable the odometer feature ;-) The used filament will be obtained by analyzing the Gcode OctoPrint sends to your printer.

Knusperhas, 2017-12-02:
Sorry for late answer. Now i can say: Thank you for the great tool :) (It's just a little bit confusing with the Odometer without odometer feature, maybe it would be better with another name... as example "use Gcode for length" )
If i take a new Role and type in the same weight in remaining weight it works verry fine. 
But if i choose an old role and type in a lower remaining weight it is not counting down this weight... 
but it's not so bad.

malnvenshorn, 2017-12-02:
Well, an odometer is a tool for measuring the distance therefore I thought the name fits. But I'm considering to rename it if it leads to confusion.

> But if i choose an old role and type in a lower remaining weight it is not counting down this weight...

Feel free to open a bug ticket if you encounter this issue.

closed 2017-12-02 by malnvenshorn