malnvenshorn/OctoPrint-FilamentManager #50: update to 0.5.0 fails

sticilface, 2017-12-12:
Saw you rolled the update into 0.5.0.  Trying to update through the GUI...

```
Updating, please wait.
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.0.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.0.zip (235kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.0)
Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.0)
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.0)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.0)
Using cached psycopg2-2.7.3.2.tar.gz
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-YWr_EX/psycopg2/
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
----------------------------------------
The update did not finish successfully. Please consult the log for details.
```

any clues?

malnvenshorn, 2017-12-12:
Try `apt-get install libpq5` on raspbian/debian or respectively `pacman -Sy postgresql-libs` if you're running arch linux.

Angelinsky7, 2017-12-12:
Hi,
Sadly, after updating, upgrading, installing libpq5 and rebooting the error is still there

Angelinsky7, 2017-12-12:
```
apt-get remove libpq5
apt-get install libpq-dev
```

did the trick...
it's sad that those libs aren't installed automatically with the gui...

malnvenshorn, 2017-12-12:
Added the required dependencies to the README and the release notes.

malnvenshorn, 2017-12-12:
> it's sad that those libs aren't installed automatically with the gui...

Sadly I cannot ship this with the release. There exists a `libpq-dev` python package but it needs to be installed with the same version as PostgreSQL and I cannot predict the PostgreSQL version provided by your distribution. If someone knows a solution please tell me.

JD2005L, 2017-12-13:
I'm getting the following errors when I try and install libpq-dev:

```
pi@octopi:~ $ sudo apt-get install libpq-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  comerr-dev krb5-multidev libgssrpc4 libkadm5clnt-mit9 libkadm5srv-mit9 libkdb5-7 libpq5
Suggested packages:
  doc-base krb5-doc krb5-user postgresql-doc-9.4
The following NEW packages will be installed:
  comerr-dev krb5-multidev libgssrpc4 libkadm5clnt-mit9 libkadm5srv-mit9 libkdb5-7 libpq-dev libpq5
0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 428 kB/731 kB of archives.
After this operation, 2,974 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main libgssrpc4 armhf 1.12.1+dfsg-19+deb8u2
  404  Not Found [IP: 93.93.128.193 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main libkadm5clnt-mit9 armhf 1.12.1+dfsg-19+deb8u2
  404  Not Found [IP: 93.93.128.193 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main libkdb5-7 armhf 1.12.1+dfsg-19+deb8u2
  404  Not Found [IP: 93.93.128.193 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main libkadm5srv-mit9 armhf 1.12.1+dfsg-19+deb8u2
  404  Not Found [IP: 93.93.128.193 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main krb5-multidev armhf 1.12.1+dfsg-19+deb8u2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/k/krb5/libgssrpc4_1.12.1+dfsg-19+deb8u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/k/krb5/libkadm5clnt-mit9_1.12.1+dfsg-19+deb8u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/k/krb5/libkdb5-7_1.12.1+dfsg-19+deb8u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/k/krb5/libkadm5srv-mit9_1.12.1+dfsg-19+deb8u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/k/krb5/krb5-multidev_1.12.1+dfsg-19+deb8u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Angelinsky7, 2017-12-13:
did you 

```
apt-get update
apt-get dist-upgrade -y
apt-get upgrade -y
```

before ??

malnvenshorn, 2017-12-13:
@Angelinsky7 Are you referring to me? I've no debian (based) distribution running.

Angelinsky7, 2017-12-13:
@malnvenshorn no sorry i received a mail from that thread but it didn't appear here... (???)
so i answered here before noticing that it was a direct question....

Angelinsky7, 2017-12-13:
@malnvenshorn thanks anyway for your help :-)

bryanhunwardsen, 2018-01-11:
Not sure what the library is being leveraged for, thought Id add that a great deal many people run octoprint off of raspian/octopi, not many will be able to trace error here or manually fix as detailed. Would suggest finding alternate implementation removing this dependency(I have 20+ plugins installed and another 15-20 that i tried and uninstalled for various reasons) none of which was not manageable through the gui. This is such a great plugin, I only wish it to be useable by all! In the meantime, Id humbly suggest explicitly capturing this error and tailing the failure with a verbose human log statement that points directly to the readme and the issue detailed in a simple format to it is easilly understood and recoverable by perople who may not be comfortable in github or at an ssh terminal. regards, hope you can find a long term solution to support ui only management.

malnvenshorn, 2018-01-11:
The library is required because the Python module for the PostgreSQL support uses the `libpq` client library. I totally agree with you that this is not the ideal situation now.

I cannot capture this error because it appears during the installation process (at least I don't know how), but even if it were possible I don't think it would make a difference. The amount of tickets showed that many people don't read the output or the log files. 

I removed the `psycopg2` module, required for the PostgreSQL support, from the dependencies for the plugin version. This shifts the "issue" only to the people who are using PostgreSQL as their database. They have to install this module manually.

closed 2018-01-11 by malnvenshorn