malnvenshorn/OctoPrint-FilamentManager #51: Install 0.50 fail with error code 1 in /temp/pip-build-KDVgMv/psycopg2/

maan28, 2017-12-13:
Install 0.50 with OctoPrint 1.3.6 (new) fail. Here the ErrorMessage:

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-KDVgMv/psycopg2/
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Fehler!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

compmas2, 2017-12-14:
I am having the same issue.  Here is my information:

#### What were you doing?

I am running OctoPrint on a Raspberry Pi 3 running the newest version of Raspbian.  I upgraded OctoPrint from 1.3.5 to 1.3.6 using the web GUI.  At the same time I attempted to upgrade a few plugins which included updating OctoPrint-FilamentManager from 0.4.1 to 0.5.1.

Attempts to update OctoPrint-FilamentManager from 0.4.1 to 0.5.1 on its own through the web GUI also failed.

#### What did you expect to happen?
I expected the update of OctoPrint-FilamentManager to succeed without an error.

#### What happened instead?
The OctoPrint-FilamentManager update failed.

#### Link to octoprint.log

https://www.dropbox.com/s/ck8m0ea58vri7ps/octoprint.log?dl=0

#### Link to gcode file

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Further information

At first when attempting to upgrade the plugin, it failed with an error: 
 Command "python setup.py egg_info" failed with error code 1.  This was for the gitsome package.  I resolved this by running the following command from the command line:  sudo -H pip3 install gitsome.

Then I tried the update of the OctoPrint-FilamentManager plugin to 0.51 again through the web GUI and got the following error:

```Updating, please wait.
++++++++++++++++++++++++++++++++++++++
Now updating Filament Manager to 0.5.1
++++++++++++++++++++++++++++++++++++++
/home/pi/OctoPrint/venv/bin/python -m pip install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip (233kB)
Requirement already satisfied: OctoPrint in /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.1)
Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.1)
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.1)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.1)
Using cached psycopg2-2.7.3.2.tar.gz
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-SelUTI/psycopg2/
The update did not finish successfully. Please consult the log for details.
```

I have read the FAQ and provided all required information.

compmas2, 2017-12-14:
I just looked at another ticket (https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/50) and found the following command that fixed my issue:

apt-get install libpq-dev

closed 2017-12-14 by malnvenshorn