malnvenshorn/OctoPrint-FilamentManager #54: Command "python setup.py egg_info" failed with error code 1

Vikinge, 2017-12-15:
Error updating to 0.5.1
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-CSvRKt/psycopg2/

Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-CSvRKt/psycopg2/
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
The update did not finish successfully. Please consult the log for details.

Vikinge, 2017-12-15:
What were you doing?
Updating with Software Update Webui

What did you expect to happen?
update to complete and restart

What happened instead?
Update Failed.

Link to octoprint.log
previous posting

Link to contents of Javascript console in the browser

Further information
Update Failed!  The update did not finish successfully.  Please consult the log for details.
Restarted the PI, same error.  Also tried on a second OctoPI, same exact errors.  I did apt-get update, rebooted and still same error.

malnvenshorn, 2017-12-15:
You will find the solution by reading the release notes

closed 2017-12-15 by malnvenshorn

Vikinge, 2017-12-15:
I see.  Thank you.
apt-get install libpq-dev