malnvenshorn/OctoPrint-FilamentManager #56: add rfid support?

dwc56, 2017-12-16:
could you add frid support thru gpio pins so you would scan a rfid card and then it would switch to a difrent spool or have a simple com system so pin 1 would be spool 1 pin 2 would be pin 2 or uart or something like that?

malnvenshorn, 2018-01-10:
It's possible to add support for NFC to automatically select the current spool, but I don't have any NFC board for the raspberry pi for testing. The next issue is which NFC chip I should support. After a quick search I found that the most common modules uses the RC522, PN532 or PN512. Each of them require a different library. This requires more research to find the best suited for the job. 

This feature might come some day, but I don't think that I will implement this in the near future.

svh1985, 2018-05-23:
Would love to have this as well. Maybe I can test it, I have some Ultimaker NFC spools, a spare RPi and nfc reader. 

Found these posts about NFC:
https://stm32f4-discovery.net/2014/07/library-23-read-rfid-tag-mfrc522-stm32f4xx-devices/
https://community.ultimaker.com/topic/15800-is-the-format-for-the-spool-rdid-tag-available/

The cheap NFC reader I have and used in the posts:
[image]

malnvenshorn, 2018-05-24:
I have to admit this feature is not at the top of my to-do list and I also doubt weather it is a useful addition. Let me explain why. 

It would have an added value if you can just put your spool in your holder and the plugin would do the rest (reading the tag, selecting spool, etc). But this is not as simple as it sounds. First you have to integrate the reader into your spool holder, second the nfc tag has to be at a fixed position, so the plugin can read the tag in constant intervals to automatically detect a filament change. The real problem is with the second requirement, because the spool is spinning :laughing: 

I know the solution from the [Da Vinci Junior](https://www.computerworld.com/article/2910250/review-the-da-vinci-junior-is-the-easy-bake-oven-of-3d-printers.html) where you have an extra tag holder (see the link). The downside is you need a separate holder for every spool, otherwise you have to disassemble it, change the tag inside and reassemble it. Manually selecting the spool would be much easier in this case. You also need different holders for different spool sizes. This doesn't sound very user friendly for me.

I'm not sure how it is solved with the Ultimaker 3. My guess is that they also have an reader inside the spool holder, but it looks like they have multiple tags inside the spool?!

As always comments are welcome :wink:

svh1985, 2018-05-24:
Although it would be nice if we can create a similar system as UM or the Da Vinci solution, I think we can create something very similar without embedding tags and have custom circular NFC readers.

I would suggest to pair the NFC tag and a filament rol by keeping them together (in the box, while on the printer). When you change the filament you simply take the NFC tag out of the box, scan it, load the filament and your all set. When the filament runs out... you just reassign the NFC tag to a new filament roll in Filament Manager and your good to go.

If someone thinks of a better way, to scan the NFC chip automatically, that would be great... we're all makers, right?!

malnvenshorn, 2018-05-25:
> I would suggest to pair the NFC tag and a filament rol by keeping them together (in the box or while on the printer). When you change the filament you simply take the NFC tag out of the box, scan it, load the filament and your all set. When the filament runs out... you just reassign the NFC tag to a new filament roll in Filament Manager and your good to go.

Hm, what's the advantage over selecting the filament within the web interface?

dwc56, 2018-05-25:
you could also attach the tag to the side of the spool and the reader to the side of the spool holder so it would scan it automatically

malnvenshorn, 2018-05-25:
Yes, but it would only work with spool holders like [this one](https://www.thingiverse.com/thing:2047554) and you need a cover for the hole too. Personally I don't like those, because they tend to slide if the spool looses weight.

malnvenshorn, 2019-02-28:
After a lot of time I want to revive this request. The question is whether this feature would be just a gimmick or would it add some value to this plugin, e.g. by making it easier to select a spool and not forgetting about it.

malnvenshorn, 2019-03-02:
Another important problematic comes to my mind if you have multiple extruders. The suggestion from @svh1985 won't work in this case because how do you select the extruder you want assign the spool to?

ruedigernarr, 2019-10-05:
Using a simple usb barcode scanner could solve the same issue more simple maybe as reported here https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/177

USB barcode readers usually work plug and play and are detected as keyboards. Also most spools have already barcode labels attached which could be used as IDs, so printing is obsolete in many cases. 

Its also easier for other users to adapt than getting into rfid tagging I believe. I have both rfid and qr readers in the office but would prefer QR because its just much simpler and less buggy.

ruedigernarr, 2019-10-05:
Barcodes would also work with multiple extruders, you can add barcode labels to the printer which act as extruder prefixes to the spool. Meaning you would scan the extruder barcode then the spool and so on for the next spools.