malnvenshorn/OctoPrint-FilamentManager #57: Error - sudo apt-get install libpq-dev

simonduz, 2017-12-19:
#### What were you doing?
running sudo apt-get install libpq-dev  on a new install of octopi 1.3.6
#### What did you expect to happen?
Install the package libpq-dev
#### What happened instead?
Multiple errors relating to path for library files.  All of them displayed an error.
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main li.............................................................

#### Further information
This was a brand new install of octopi 1.3.6.   This was the first plugin I have attempted to install.
I was following the FAQ trying to install the dependencies when I got these errors.
I have not gone any further with the install.

I have read the FAQ and provided all required information.

gretel, 2017-12-19:
ya, same here. solved via https://stackoverflow.com/questions/11618898/pg-config-executable-not-found

malnvenshorn, 2017-12-22:
Sorry, but I cannot provide support for distribution specific issues => closing

closed 2017-12-22 by malnvenshorn

malnvenshorn, 2017-12-22:
As I understood it @simonduz has an issue installing the required `libpq-dev` package. This issue is not related to the plugin.

The mentioned package is needed for the new postgresql support.