malnvenshorn/OctoPrint-FilamentManager #58: Update to 0.5.1 failed

Stady234, 2017-12-21:
Updating, please wait.
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.1)
Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.1)
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.1)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.1)
Using cached psycopg2-2.7.3.2.tar.gz
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-4ggZQD/psycopg2/
The update did not finish successfully. Please consult the log for details.

Stady234, 2017-12-21:
What were you doing?
[Trying to update]

What did you expect to happen?
[to work]
What happened instead?
[Updating, please wait.
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.1)
Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.1)
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.1)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.1)
Using cached psycopg2-2.7.3.2.tar.gz
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.]

Link to gcode file
[N/A]

Link to contents of Javascript console in the browser
[N/A]

Further information
[None]

I have read the FAQ and provided all required information.

malnvenshorn, 2017-12-22:
You should read the release notes / readme to find out the additional dependencies you have to install.

closed 2017-12-22 by malnvenshorn