malnvenshorn/OctoPrint-FilamentManager #60: update failed now clean install fails

mitchcecg, 2017-12-21:
I was updating from ver 4x to yhe current version it kept failing so i uninstalled ver 4.x and now I get this error
Error!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

the 2nt time I tried it I get 
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-yblmUM/psycopg2/Error!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Any help???

zion2k, 2017-12-23:
Hello, after updating OctoPi to OctoPrint v1.3.6 I tried to update the Filament Manager plugin from 0.4.0 to 0.5.1 and got the following error.

> Aktualisiere gerade, bitte warten.
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip (233kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.1)
Collecting backports.csv<1.1,>=1.0.5 (from OctoPrint-FilamentManager==0.5.1)
Using cached backports.csv-1.0.5-py2.py3-none-any.whl
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.1)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.1)
Using cached psycopg2-2.7.3.2.tar.gz
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-kvm8Qi/psycopg2/
Die Aktualisierung wurde nicht erfolgreich abgeschlossen. Bitte konsultiere das Log für Details.

TylonHH, 2017-12-26:
same here
```writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found

Error: pg_config executable not found.

Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:

python setup.py build_ext --pg-config /path/to/pg_config build ...

or with the pg_config option in 'setup.cfg'.

----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-MAGwsj/psycopg2/```

sennalb, 2017-12-27:
Also I have the same problem installing this plugin on Octorpint 1.3.6 on Xubuntu 17.10
thanks
Alberto

floflogreco, 2017-12-28:
I had the same issue, just read the installation notes and it worked.
so SSH your pi and run this command 
`sudo apt-get update
sudo apt-get install libpq-dev`

closed 2017-12-29 by malnvenshorn