malnvenshorn/OctoPrint-FilamentManager #62: Option to force spool select before start printing

gkrystev, 2017-12-26:
It will be very useful if there is an option which forces the user to select loaded spool before start printing. I saw there is a feature request for RFID, which will automate the spool selection, but for manual mode, the force spool select will be useful, because from my experience, many times I just hit print and forgot to select the spool which is loaded on the printer.

malnvenshorn, 2017-12-27:
Have you tried the extra confirmation dialog which can be enabled in the settings?

gkrystev, 2018-01-08:
Did not saw this option till now. 

However, when the option is enabled, I'm getting a dialog that I need to confirm a spool, with only "Cancel" and "Start Print" button,  but the "Start Print" button is always grayed out and can't continue. Honestly I expected to see something like the selected spool if any was selected and if no spool was selected, to be able to select one and continue.

Also, this dialog appears only when starting the print with the  "Print" button, but if I'm using the "Load and Print" button (located by each file) there is no any warning/conformation dialog at all.

malnvenshorn, 2018-01-09:
> However, when the option is enabled, I'm getting a dialog that I need to confirm a spool, with only "Cancel" and "Start Print" button, but the "Start Print" button is always grayed out and can't continue. 

You have to select the same spool you have selected in the sidebar.

> Honestly I expected to see something like the selected spool if any was selected and if no spool was selected, to be able to select one and continue.

You can read #36 if you want to know why it is designed how it is.

malnvenshorn, 2018-01-10:
> Also, this dialog appears only when starting the print with the "Print" button, but if I'm using the "Load and Print" button (located by each file) there is no any warning/conformation dialog at all.

Fixed on the devel branch. The confirmation dialog also appears now when using "Load and Print".

closed 2018-01-11 by malnvenshorn