malnvenshorn/OctoPrint-FilamentManager #65: [Request] add Filament spool weight

TylonHH, 2018-01-01:
Hi,
what do you think if I weight the hole spool and give the known spool weight too?

Cheers

malnvenshorn, 2018-01-09:
Sorry, but I don't know what you mean.

TylonHH, 2018-01-09:
Sorry, I described it so confusion.

I mean, I know the hole weight of the spool, because I can weight it.
Must manufactures give the weight of the empty spool too. Or I measure an empty spool.
Because your plugin needs to know just the weight of the filament.

Filament [g] + empty spool [g] = hole spool [g]
[g]=[gramm]

TylonHH, 2018-01-09:
Oder einfach gesagt: wäre es nicht einfacher wenn ich das Gewicht der einzelnen leeren Spule angebe? Dann brauche ich nur einmal die ganze Spule mit Filament wiegen und er Spulenkörper wird abgezogen damit man das Gewicht des reinen Filament hat.

malnvenshorn, 2018-01-09:
> Because your plugin needs to know just the weight of the filament.

Exactly. So why I need the weight of the empty spool? You enter the weight of the filament in the spool properties. There is no need for any calculation.

TylonHH, 2018-01-10:
There is an need for calculation.
I just can weight the hole spool with filament. Then I have to research the weight of the empty spool. But if the weight of an empty spool is in my/your database I could just enter the weight of the hole spool.
You know what I mean?

malnvenshorn, 2018-01-10:
The only case in which you might need to weight your spool is when you want to add an already used spool. I don't see why storing the empty spool weight is necessary if you need this value only once.

TylonHH, 2018-01-10:
Ahhh, know I get your point of view. In this case I totally agree.
That's why I came up with my request, I'm inventory all my used(!) spool. But you're right there is no really for need of this when u do this immediately after I open a new spool.

closed 2018-01-10 by TylonHH