malnvenshorn/OctoPrint-FilamentManager #66: Upgrade Error

Hobb3s, 2018-01-03:
Hi,
I'm attempting to upgrade from Filament Manager 0.4.1 to the latest and I get an error during the upgrade in Octoprint 1.3.6. 

**Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-U0W4HK/psycopg2/**

Here's the log output from the install attempt:

>/home/pi/oprint/bin/python -m pip install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip (233kB)
Requirement already satisfied (use --upgrade to upgrade): OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.1)
Requirement already satisfied (use --upgrade to upgrade): backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.1)
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.1)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.1)
Using cached psycopg2-2.7.3.2.tar.gz
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-U0W4HK/psycopg2/
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.

sajovicd, 2018-01-04:
Hello, I have the same problem.

modidep, 2018-01-05:
me 2
Updating, please wait.
++++++++++++++++++++++++++++++++++++++
Now updating Filament Manager to 0.5.1
++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python -m pip install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip (233kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.1)
Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.1)
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.1)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.1)
Using cached psycopg2-2.7.3.2.tar.gz
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-ZmCH0M/psycopg2/

modidep, 2018-01-05:
me 2

CarnufexG, 2018-01-07:
One more me too!

closed 2018-01-08 by malnvenshorn