malnvenshorn/OctoPrint-FilamentManager #67: Unable to install the Filament Manager

CADMunkie, 2018-01-04:
Before opening a new ticket please take a look if there is already an existing one addressing the same concern.

Also read the FAQ: https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki#faq

If you have a **request** or **question** mark them with a "[Request]" respectively "[Question]" prefix in the title.

For **bug** reports use the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the release notes.

(Before submitting your ticket, please delete this text up to and including the line too - it's only here for your information)

---

#### What were you doing?

I am not able to install the app 

writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-14r6Wa/psycopg2/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

#### What did you expect to happen?

Install from the Plug In Manager

#### What happened instead?

#### Link to octoprint.log

Installing plugin "FilamentManager" from https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip...
/home/pi/oprint/bin/python -m pip install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip (233kB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.1)
Collecting backports.csv<1.1,>=1.0.5 (from OctoPrint-FilamentManager==0.5.1)
Using cached backports.csv-1.0.5-py2.py3-none-any.whl
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.1)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.1)
Using cached psycopg2-2.7.3.2.tar.gz
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
warning: manifest_maker: standard file '-c' not found
Error: pg_config executable not found.
Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:
python setup.py build_ext --pg-config /path/to/pg_config build ...
or with the pg_config option in 'setup.cfg'.
----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-14r6Wa/psycopg2/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

#### Link to gcode file

[If you encounter issues with the odometer feature provide a small gcode file which produces the problem.]

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

[If you have more information which could help me to understand/fix the issue feel free to add them here.]

I have read the FAQ and provided all required information.

TomW1605, 2018-01-05:
i have this problem as well is there a fix?

TomW1605, 2018-01-05:
i have installed 0.4.1 as an interum

sameSEA, 2018-01-06:
I think this is not an issue and should be closed.
Should be installed according Setup procedure.

TomW1605, 2018-01-06:
@sameSEA except that using the setup procedure doesn't work

sameSEA, 2018-01-06:
Well, I was getting same error and log output.
Then I installed libpq-dev according setup section on main page. And installation went properly

TomW1605, 2018-01-06:
oh those instructions
i was unable to install it
apt-get was giving me an error

sajovicd, 2018-01-06:
I have same error.
I think that if I install plugin thrue plugin manager, it must install automagically everything that is needed.
If it has to be installed manualy than it shuld not be possible to install thrue plugin manager.

CKENT0369, 2018-01-06:
Cannot install the required dependency either. I also agree that it should pull the require dependency from the install manager in octopi.  Getting a buch of "Failed to fetch" when trying to install dependency from ssh. I installed 4.1 and it works fine.

PreownedFIN, 2018-01-07:
I have this same problem too. I'm trying to update through the plugin manager and it throws error about pg_config not being found.

Since the plugin manager can't install this update, this is undoubtably a bug somewhere. From the error message I'd guess that something wasn't packaged that should've been packaged...

```
++++++++++++++++++++++++++++++++++++++
Now updating Filament Manager to 0.5.1
++++++++++++++++++++++++++++++++++++++
/home/pi/OctoPrint/venv/bin/python -m pip install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Collecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip
Downloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/0.5.1.zip (233kB)
Requirement already satisfied: OctoPrint in /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg (from OctoPrint-FilamentManager==0.5.1)
Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.1)
Collecting SQLAlchemy<1.2,>=1.1.15 (from OctoPrint-FilamentManager==0.5.1)
Using cached SQLAlchemy-1.1.15.tar.gz
Collecting psycopg2<2.8,>=2.7.3 (from OctoPrint-FilamentManager==0.5.1)
Using cached psycopg2-2.7.3.2.tar.gz
Complete output from command python setup.py egg_info:
running egg_info
creating pip-egg-info/psycopg2.egg-info
writing pip-egg-info/psycopg2.egg-info/PKG-INFO
writing top-level names to pip-egg-info/psycopg2.egg-info/top_level.txt
writing dependency_links to pip-egg-info/psycopg2.egg-info/dependency_links.txt
writing manifest file 'pip-egg-info/psycopg2.egg-info/SOURCES.txt'
Error: pg_config executable not found.

Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:

python setup.py build_ext --pg-config /path/to/pg_config build ...

or with the pg_config option in 'setup.cfg'.

----------------------------------------

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-OmkLCU/psycopg2/
```

malnvenshorn, 2018-01-08:
@sajovicd It's not possible to ship the dependency with the plugin. Of cause I can remove this plugin from OctoPrints' plugin repository, but it will make it hard for users to find this plugin. To be honest I wouldn't care because I wrote this plugin for myself, but I think some people out there might find it useful as well. That's why I put it on github in the first place.

For everyone who has this problem. Reading the release notes, readme or looking at the open tickets would have solved your issue very quickly. It's quite annoying that a new issue is opened after another...

I might find another solution if I have some time to look into this. But for now it is what it is.

closed 2018-01-08 by malnvenshorn

sajovicd, 2018-01-08:
I did manage to instal it after apt-get update, apt-get upgrade the system. Then everything went smooth.
Thanks