malnvenshorn/OctoPrint-FilamentManager #68: [Request] Display temperature offset in list view

fedhat, 2018-01-06:
The temperature offset feature is very useful. I'd like to request that the offset value appear in its own column in the Filament Spools table view, perhaps between the Vendor and Weight columns.
[image]

malnvenshorn, 2018-01-09:
Hm... is there are reason why you want a column for the temperature offset? I, personally, don't see the benefit of adding this value to the table. I'm asking because the table width is limited and this means for each additional column a need to shrink/remove other columns.

fedhat, 2018-01-10:
I suppose it depends on how useful you think the offset is and perhaps you envisioned using offsets differently than I do. My primary use case for the offset is that I review it whenever I print something to make sure that the final temperature is correct for the filament I'm using. For example, this allows me to set all my gcodes to 190C, but then have offsets for each spool.

Displaying the temperature offset is useful to me because then I don't have to remember what offset, if any, I added to a spool. Currently, before I print something, I need to go into Filament Manager and click on the edit button to see the offset value.

If space is a concern (though I think there's enough room for an Offset column) perhaps you could remove the Used column, since you already display Weight and Remaining.

In fact, in the side bar, the Filament Manger dropdown displays the name of filament, remaining weight, type and manufacturer. In my opinion, manufacturer is much less useful than offset and I would request swapping those values as well in the selection dropdown.

malnvenshorn, 2018-01-10:
I understand your point, but why do you have to go to the spool dialog to see the offset value? When you select a spool in the sidebar the temperature offset gets applied automatically and you can see the value in the temperature tab.

The displayed properties  in the sidebar are chosen to maximize the distinguishability of your spools. In this aspect I consider the vendor more suited than the temperature offset.

fedhat, 2018-01-10:
I didn't know that the offset field under the temperature tab populated when you selected a filament from the Manager, so thank you for that tip. I'd still argue that is not good UX (controls and their effects should be in the same visual block), but now that I know where to look on screen, I can get around that particular issue.

That said, adding an offset column to the Filament Spools table, which was the initial request, helps users manage their spools. I don't see the downside of adding that information there. We disagree about whether there is room for the column, though I see gaps between Vendor / Weight and Remaining / Used where a thin column would fit without issue. The gaps remain even when I shrink the window down below the standard 975px for the large modal (`modal.large {width: 975px; margin-left: -487px;`}. The gaps are also apparent on an iPad or iPhone X in landscape mode (see attached images). The only way to get the gaps to disappear is to either shrink the browser way down or to view the Filament Manager on a mobile phone in portrait. What screen size in particular are you concerned about? 

[image]
[image]
[image]

malnvenshorn, 2018-01-10:
Yes there might be a gap between the vendor and weight in your case, but this doesn't mean that this is the case for everyone. I tried to provide the best readability with the column weights I chose. Which means that (hopefully) nothing gets cut off.

Now you could argue that there is also a gap between remaining and used, but if you switch to the german translation it will look different.

As I said before, the table width is limited. That's the way it is.

fedhat, 2018-01-10:
Ok, well thanks for taking the time to discuss it. I'll close the request.

closed 2018-01-10 by fedhat