malnvenshorn/OctoPrint-FilamentManager #70: [Request] Move filament database to tabs at top.

MStylesMS, 2018-01-12:
Very nice plug-in!  Thank you for creating it.  

One feature request is to move the management of the spools to a tab at the top of the screen rather than it being hidden down in the settings.

malnvenshorn, 2018-01-12:
The main reason why the spools table locates in the settings is that there is more space. The width of the tab view on the main screen is significantly smaller. Sometimes there are requests for additional table columns which I have to decline because there is not enough space for that.

I'd also like to move this part to a tab view because it has nothing to do with settings. But as you see I cannot just move it as it is now. Maybe if I get an idea on how to redesign the spools table/management I'm moving this part.

MStylesMS, 2018-01-15:
That makes sense and I've been pondering it a bit.  The limited width is a limitation.  ;-)  So, thinking about this a bit more today had a thought I thought I'd throw over to you.  (Sorry...in a punny mood tonight.)  

One approach that came to mind was to take a hybrid approach that combines your current table with the file list approach in the left pane of OctoPrint.  To me, the main three things to see are 1) Name, 2) Material, and 3) Remaining.  The rest is interesting, but I don't need that as much.  So here are the thoughts:

1)  I'd try to keep those as three columns similar to what you have now.  
2) I'd also keep the Action column but add in the down carrot (or maybe an information icon a "!" inside a circle).  When you click on that then you could show all kinds of detail for the record.  

I realize there others might want some other details shown where you could sort or scan down a column, but I'm not sure why most of those things would be that important.

reven, 2018-04-11:
I think that @MStylesMS suggestion woould be a great compromise. Just to see the inventory information with the basic info and get a *bird's view* of your inventory would be great. I would very much like to see that in a tab at the top.

malnvenshorn, 2018-04-25:
I'm currently thinking about adding this request to the 0.6.0 milestone. 

Here some idea how it might look like:
[image]

I totally got rid of the total weight and used column, because I agree that they are not that important. But I consider to add some filter options, e.g. for the material type.

reven, 2018-04-25:
That looks great! Can't wait. 😃

malnvenshorn, 2018-04-26:
An update. As always your feedback is welcome :wink: 

[image]

malnvenshorn, 2018-04-27:
If someone wants to try it out, this feature is now available on the `feature/move-inventory-to-tab` branch. Everything except the filtering should work.

reven, 2018-04-27:
Sven, this looks really great. I've tried duplicating and editing spools and everything seems to work. It's great to have all the info there as a tab and not have to go into the settings. I'll try printing a few things and see how it holds up. I'll let you know if anything breaks.

Thanks for the hard work!

MStylesMS, 2018-04-27:
I’m traveling but will give it a try next week.  Thanks!!!

Mark Stevens

All typos are Siri's fault!

malnvenshorn, 2018-04-29:
I came across some issue. If you have a spool selected and apply a filter which doesn't include the selected spool the selection will be reset.

Two solutions come to my mind:
1. I can make the selection independent from the inventory table. That means sorting and filtering will not be applied.
1. If you think it would be cool to keep that dependency I have to make sure that the selected spools don't get removed from the selection (or even the inventory?). The question  in this case is how do I highlight them to show that they are actually not part of the filtered entries. 

Which solution would you prefer?

mickdekkers, 2018-04-30:
@malnvenshorn what is selection in this case? Are you referring to clicking on the dropdown menu to the right of the spool, where having that open means the spool is "selected"? Or are we talking about selection in the more traditional sense, where you click on one or more items in the table to highlight them and later perform some action on the selection?

If it's the latter, let's see...

- If you keep the selection visible in the table when the user changes the sorting or filtering options, they may wonder why some entries are not filtered out. As you've implied, this can be unintuitive without somehow clearly conveying to the user why this is happening.
- If you clear the selection when the user changes the sorting or filtering options, this may be unexpected.

A third, more complex option would be to copy what Google does with (desktop) Gmail. 

- The selection is independent from the table, so changing the filter retains your selection "in the background".
- The actions you perform (e.g. delete, archive) only affect the visible part of the selection, so only the emails that are:
  - selected
  - match the filter (if any)
  - visible on the current page

malnvenshorn, 2018-04-30:
I'm talking about the spool selection list in the sidebar. At the moment it uses the same data source as the inventory table which causes the described behavior.

mickdekkers, 2018-04-30:
Oh, I see. I think it would be a good idea to keep that selection list simple. It's a quick way to pick from all your available spools, and I'm not sure whether filtering would add much value to it. So making it independent from the inventory table would probably be best.

Here's a thought: how about adding a button to the spools on the inventory table to activate them? This would still allow users to apply filtering and sorting to narrow the inventory table down to the spool they're looking for, and you can avoid the dependency issue between the selection list and the inventory table.

Here's a quick photoshop:
[image]

malnvenshorn, 2018-05-01:
Thanks. This as a great idea!

malnvenshorn, 2018-05-01:
Few changes:
- Page size will be restored on reload
- Spools can also be selected from the inventory table
- Sidebar selection is not affected by sorting and filtering of the inventory table

malnvenshorn, 2018-05-03:
Some last minor changes:
* Disabling certain inventory actions if user is not logged in
* Automatically apply filter if target column is changed
* Reset button for inventory filter

I think that's it, unless someone stumbles across issue :wink:

szafran81, 2018-12-31:
+1
Great work. Thank you.

ChoccyHobNob, 2019-10-03:
I was just going to suggest this idea! Great to see it's already been done, any idea when it will go live?

Silventar, 2020-02-13:
Love this plugin; so useful. This new tabs feature I was dissapointed to see was not live yet in the main branch...any idea when it will be included?

schuessm, 2020-05-05:
I also looking for the TAB feature. It will be great if you develop it or launch. great plugin I like it

piotr-kubica, 2023-08-06:
How can I enable this TAB feature?