malnvenshorn/OctoPrint-FilamentManager #72: Failed to update filament on tool0 - Port issue?

TylonHH, 2018-01-14:
Hi,
Maybe I'm wrong but I guess its because I run OctoPrint on Port 88. Could that be an Issue?
How can I solve this?

TylonHH, 2018-01-14:
Or maybe this?
```
pi@octopi:~ $ python -V
Python 2.7.9
```

TylonHH, 2018-01-14:
Here is the full Log aftern an clean instal of your Log
[LOG.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1629318/LOG.txt)

malnvenshorn, 2018-01-14:
Please follow the issue template and provide *ALL* information. Especially the first question is important because I would guess that your database might be corrupt.

TylonHH, 2018-01-14:
#### What were you doing?
I installed earlier an older Version. But the weight wear never substracted. So I exported the the profiles and spool csv and did an uninstall of this plugin. I also deleted manually  ``/home/pi/.octoprint/data/filamentmanager/filament.db``.
Then I installed this plugin again. I couldn't import the zip neither the weight is substracted

#### What did you expect to happen?
A (correct) calculation of the left filament

#### What happened instead?
No calculation, even aftern an clean install

#### Link to octoprint.log
[LOG.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1629629/LOG.txt)

#### Link to gcode file
I guess its no Issue with gcode but here is an example. Sliced with Simplify3D 4.0.1
[heattower_medium.gcode.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1629630/heattower_medium.gcode.txt)

#### Link to contents of Javascript console in the browser
[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information
I'm running OctoPrint on Port 88

I have read the FAQ and provided all required information.

TylonHH, 2018-01-15:
I deleted all entries related to FilamentManger in the config.yaml
Now this is my new config.yaml 
[config.yaml.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1632671/config.yaml.txt)
I also deleted the database and uninstalled the Plugin. Rebooted the hole Pi an did an fresh clean install.
Now your plugin doesnt appear in the menu of OctiPrint, so I cant set it up.
Here is the Log
[LOG.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1632675/LOG.txt)

Here is my older config.yaml
[OLD_config.yaml.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1632677/OLD_config.yaml.txt)

I gues now I delted to much from the config?

malnvenshorn, 2018-01-15:
Not at all, but for some reason OctoPrint doesn't find/register the filament manager plugin. :confused:  Maybe reinstalling helps?!

TylonHH, 2018-01-15:
I tried twice.

TylonHH, 2018-01-15:
Here is the install Log
```

Installiere Plugin "FilamentManager" von https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip.../home/pi/oprint/bin/python -m pip install https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zipCollecting https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zipDownloading https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zipRequirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: backports.csv<1.1,>=1.0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: uritools<2.2,>=2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: SQLAlchemy<1.2,>=1.1.15 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: sockjs-tornado<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.17-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.0-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: psutil<6,>=5.4.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-5.4.1-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chain [... cut ...] rint->OctoPrint-FilamentManager==0.5.2)Requirement already satisfied: pytz>=0a in /home/pi/oprint/lib/python2.7/site-packages/pytz-2016.3-py2.7.egg (from Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-FilamentManager==0.5.2)Installing collected packages: OctoPrint-FilamentManagerRunning setup.py install for OctoPrint-FilamentManager: startedRunning setup.py install for OctoPrint-FilamentManager: finished with status 'done'Successfully installed OctoPrint-FilamentManager-0.5.2Fertig
```

malnvenshorn, 2018-01-15:
Sorry but I can't help you with this. I don't know why OctoPrint doesn't find the plugin. There is no error in the logs :confused:

TylonHH, 2018-01-15:
Now I see the Plugin, I entered a new spool, and test now.
I quick view into the Log I still see some
```
2018-01-15 21:16:34,290 - octoprint.server.preemptive_cache - INFO - ... done in 32.42s
2018-01-15 21:17:07,519 - tornado.access - WARNING - 401 GET /api/languages (127.0.0.1) 73.43ms
2018-01-15 21:17:07,574 - tornado.access - WARNING - 401 GET /api/printerprofiles (127.0.0.1) 49.04ms
2018-01-15 21:17:07,606 - tornado.access - WARNING - 401 GET /api/logs (127.0.0.1) 25.14ms
2018-01-15 21:17:07,644 - tornado.access - WARNING - 401 GET /api/settings (127.0.0.1) 30.68ms
2018-01-15 21:17:16,615 - tornado.access - WARNING - 401 POST /api/system/commands/core/restart (127.0.0.1) 55.33ms
2018-01-15 21:18:03,666 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.178.43
2018-01-15 21:18:13,283 - tornado.access - WARNING - 409 POST /api/printer/command (127.0.0.1) 54.63ms
```
May @foosel know something? I'm not sure if it's caused by the plugin

TylonHH, 2018-01-16:
After some reinstallation and reboots it works. But I dunno what was wrong.

closed 2018-01-16 by TylonHH