malnvenshorn/OctoPrint-FilamentManager #73: No selection of spool

scharrin2, 2018-01-14:
When pressing on "Print" the selection screen for the Spool is not shown and the Button "Start Print" is inactive.
When I retry 4 to 5 time the selection for the spool is shown
---

#### What were you doing?

Press on button Print (in tho top) or on the Buton "Print" above a single file.

#### What did you expect to happen?

The dropdown for the spool should be shown

#### What happened instead?

The dropdown for the Spool is missing

#### Link to octoprint.log

[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1629282/octoprint.log)

#### Link to gcode file

[If you encounter issues with the odometer feature provide a small gcode file which produces the problem.]

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

[image]

I have read the FAQ and provided all required information.

malnvenshorn, 2018-01-14:
Could you please also provide the content of the Javascript console from your browser?

scharrin2, 2018-01-14:
Hello,

Do you know how to activate on Safari browser on Mac?

Best regards,
Christian

malnvenshorn, 2018-01-14:
No sorry I don't know that. I also have to say that I'm not supporting the Safari browser because I don't own any apple device and therefore I cannot test with that browser. Maybe you could try it with  a chromium based browser or firefox.

scharrin2, 2018-01-14:
> Antworten an: malnvenshorn/OctoPrint-FilamentManager <reply@reply.github.com>

OK – I will try it with Google Chrome on Mac!

Best regards,
Christian

scharrin2, 2018-01-14:
Do you mean only the output (screenshot) from the java-script console?

It only looks like this:
[image]

malnvenshorn, 2018-01-15:
Yes that's what I meant. But I cannot see any error.

Because the dialog shows only selections for tools which are actually used by the print job I would explain this behavior with the fact that the analyzing process of the selected file is not finished at the time the dialog opens. This would correspond to the fact that the selection appears after some retries. 

It would be great if you could verify that in your case.

TylonHH, 2018-01-16:
I had this right now too, but in Firefox.
After some clicks to box was shown. My GCODE was 10MB big/small.
Unfortunately I forget the Java console. But I'll too next time.

TylonHH, 2018-01-19:
[image]

Here is the provided [Link](https://developer.mozilla.org/de/docs/Web/JavaScript/Reference/Fehler/Property_access_denied?utm_campaign=default&utm_medium=firefox-console-errors&utm_source=mozilla) in the Message above

TylonHH, 2018-01-23:
What I found out. When under "Status" the correct weight and tool is not red or calculated, your pop-up is empty. I had to refresh my browser site and/or load a different file and load the specific file again until the "Status" is correct.
I donno who is responsible for this "Status"

malnvenshorn, 2018-01-23:
Thanks. I think this confirms my previous speculation. I'm going to revise the dialog so that the user isn't confronted with an empty dialog and don't know why this happened.

TylonHH, 2018-01-23:
Yeah you're right. But please leave this popup. We should solve the problem why the status isn't updated.

sticilface, 2018-01-24:
I can confirm that I get this quite frequently too
[image]

Hitting refresh on the browser a few times gets rid of it.

malnvenshorn, 2018-01-25:
@TylonHH the only cases where this should happens are:

* The file is not analysed by OctoPrint yet
* Loading the file takes some time

If this issue appears after the used filament and print time has been updated in the sidebar then we might have a bug somewhere.

TylonHH, 2018-01-25:
@malnvenshorn you're right. That's what I ment, that someone is responsible.
I'll open an issue at @foosel GitHub.

szafran81, 2018-01-31:
It would also be nice if after that window shows up with the list of spools to select that the currently selected spool on the main screen would be preselected.

malnvenshorn, 2018-01-31:
@szafran81 see #71

malnvenshorn, 2018-02-04:
Added some information to the confirmation dialog if no selection is available to explain the cause of this.

If I get some information that show that there is a bug in the filament manager after all I'll reopen this ticket.

closed 2018-02-04 by malnvenshorn

jesasi, 2018-03-18:
I think it is bug: https://discourse.octoprint.org/t/selecting-problem-in-filamentmanager/

TylonHH, 2018-03-19:
It's not a bug of the filament manager plugin. Because this plugin don't analyze the gcode.

malnvenshorn, 2018-03-19:
Unfortunately the video from the linked  post doesn't show whether the file was fully analysed before the dialog was opened or not. Therefore I'm assuming the file was not analysed yet. In this case the behavior is as intended.

bartwybouw, 2020-05-24:
I have the same issue.
And yes the file was fully analysed.
I can select certain spools and get the start button active, others I cannot (of course the ones I need ;-) ).
What can I do to diagnose, provide loggings etc?

bartwybouw, 2020-05-24:
[image]
[image]
[image]

bartwybouw, 2020-05-24:
It seems out of my list of filaments, only one is useable.(rPLA RED)

bartwybouw, 2020-05-24:
I think I've found the issue. 
It's not a bug, but it's confusing.

You have to make sure that the filament you want to use is selected in the tool box, and matches the one that you then select in the confirmation dialog.

For me it's more an issue of not guiding the user well enough, but not a bug.

So first, select the filament you want to use in the 'tool window' of Filament manager, then hit print and confirm the SAME filament in the confirmation pop-up.

IMHO, it would be better to allow the user to switch filament in that confirmation window, but make clear that you SWITCH to a different filament. Maybe add a button "change filament" ?

[image]

dzid26, 2020-07-16:
I was confused as well. I even anticipated that this is happening..., but couldn't find option in the settings to change filament. 
 - I forgot there was the little drop list on the bottom of the main screen.