malnvenshorn/OctoPrint-FilamentManager #76: [Request] Filament change on M600

xeon927, 2018-01-17:
Hey - just wondering if it might be possible to implement a mid-print filament change for when an `M600` command is sent to printer.

I've been doing a few prints lately which have different coloured layers, and it would be awesome if it could accurately keep track of these prints (without having to manually pause print, change colour, then resume).

Thanks in advance :)

malnvenshorn, 2018-01-17:
You don't have to manually pause the print if you use the M600 command. After your print is paused change you filament, select the new one in the filament manager and continue your print. I don't know what you want me to implement :confused:

xeon927, 2018-01-18:
The main thing is that when the filament change is executed, the area to change current filament in the OctoPrint sidebar is still disabled / greyed out - hence the need to pause in order to change filament. The colour change happens just with the LCD controller (as far as OctoPrint thinks, it's still midway through a print, just a very long-running command). I was thinking something like detecting if M600 is sent, then display a dialog to prompt for what colour is being changed to, in order to properly keep track of what colour is used when.

malnvenshorn, 2018-01-18:
Ah ok, didn't knew that OctoPrint handles it that way. It might be possible to implement such thing in the filament manager. The issue is I don't have an LCD on my printer and I'm not sure if I can simulate such behavior with the virtual printer. Which means I have no way of testing it.

What you can use instead is the `M0` command which will cause OctoPrint to pause the print.

xeon927, 2018-01-18:
All right, no worries. Was hoping to use the M600 command (as that will handle moving the head out of the way, along with retracting old colour and pushing through new colour until there is no longer any reserve molten plastic in the head).

Also, if you do end up implementing something to handle this, I'm more than happy to assist with testing for you :)

brad-stone, 2018-03-19:
+1 for this improvement.

We are experiencing the exact same problem.  The M600 is sent from Octoprint to the printer, but the Octoprint controls to extrude filament (etc.) are grayed out so that you can't replace and reload the new filament.

We just need the Octoprint controls to be active when the M600 is sent and then a button to resume the print (i.e. cancel waiting).

BTW, we are loving Octoprint, so major thanks!

malnvenshorn, 2018-03-19:
I'm unsure whether I should support this command or not because you can simply use the before mentioned `M0` command or `@pause` which OctoPrint will support in the next release. Both commands will pause the print and you can change your filament.

MagicManFoli, 2018-04-02:
It seems like swapping filament also leads to the first difference being dropped, this is how I reproduced it:
1. Initialise a filament change with M600
2. Pause octoprint, this enables controls
3. Change filament in filament manager
4. Change "real" filament
5. Continue octoprint
6. Continue printer

After the print has finished the second filament is reduced, but the first filament is unchanged.

malnvenshorn, 2018-04-04:
@MagicManFoli please create a new ticket for your issue.

malnvenshorn, 2018-04-16:
I've added this request to the version 0.6.0 milestone. But this is not a promise that this feature will come. The main issue is that I haven't found a way yet to determine when a long running command has ended. This info is needed so that the filament selection can be disabled again.

malnvenshorn, 2018-04-22:
Added some basic support for the `M600` command with the commit above. It's more or less a proof of concept and available on the `feature/M600-support` branch. I'm not sure if I'm going to merge this into `master`, for example, because the confirmation dialog is not working (since the resume is not controlled by OctoPrint).

malnvenshorn, 2018-04-23:
I took up the suggestion by @xeon927 to show dialog in web interface instead of enabling the sidebar selection if a M600 command is detected. Changes available on the `feature/M600-support` branch.

xeon927, 2018-05-02:
@malnvenshorn - sorry 'bout the delay in the testing. I did a couple of tests recently - everything works perfectly!

Only thing I would probably request is to maybe have it auto pause the print, just in case an idiot like me finishes loading filament on the printer before changing it in Octoprint. However, in saying that, if the print isn't unpaused from Octoprint, the hot nozzle would be sitting at the last location over the print (which can lead to burning). Either way!

I also tested with the Octoprint window closed when the filament change starts, and it's good to see that the message still shows.

Looks good so far!

malnvenshorn, 2018-05-02:
Thought about to simply pause the print on the `M600` command too, but discarded that idea. The reason is the command issues a filament change controlled by the printer and not by OctoPrint and it should stay this way. Therefore the OctoPrint interface is only used for selecting the new filament. If someone whats to achieve the same with OctoPrint it's possible. There is the `@pause` command, start/stop scripts and it is possible to add scripts for loading/unloading the filament.

Burritobun, 2018-07-11:
Excuse my ignorance, but what would the branch specific url be to download the M600 branch onto my octoprint interface? Also does it uninstall the dependency when I remove the original branch?

Billiam, 2019-01-24:
Instead of (or in addition to) tying the filament selection to the `m600` command, could the filament select be enabled while printing?

I do use the filament swap command, but also sometimes just forget to change the current filament selection in octoprint before starting a print, and it would be nice to be able to change it instead of having that feature locked out.

bovlb, 2020-09-21:
> Instead of (or in addition to) tying the filament selection to the `m600` command, could the filament select be enabled while printing?

This would fix both the M600 problem and the "forgetting" problem.  Note that the confirmation dialog does not work if you start a print job directly from (e.g.) PrusaSlicer.