malnvenshorn/OctoPrint-FilamentManager #79: Question Temperature offset

mitchcecg, 2018-01-23:
Before opening a new ticket please take a look if there is already an existing one addressing the same concern.

Also read the FAQ: https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki#faq

If you have a **request** or **question** mark them with a "[Request]" respectively "[Question]" prefix in the title.

For **bug** reports use the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the release notes.

(Before submitting your ticket, please delete this text up to and including the line too - it's only here for your information)

---

#### What were you doing?

[Be as specific as possible. Providing a step by step guide will help me to reproduce the issue. This is needed in order to fix the problem. I cannot help you if I don't know what exactly you did to trigger the issue.]

#### What did you expect to happen?

#### What happened instead?

#### Link to octoprint.log

[Always include the log file.]

#### Link to gcode file

[If you encounter issues with the odometer feature provide a small gcode file which produces the problem.]

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

[If you have more information which could help me to understand/fix the issue feel free to add them here.]

I have read the FAQ and provided all required information.

malnvenshorn, 2018-01-25:
It sets the temperature offset for the hotend.

closed 2018-01-25 by malnvenshorn

mitchcecg, 2018-01-25:
Please define what you mean by Offset is that a difference between what the setting is in the slicer or is that the actual temperature for the slicer

THANX
MITCH

Closed #79<https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/79>.