malnvenshorn/OctoPrint-FilamentManager #82: [Request] Support for External Database

bleem313, 2018-02-17:
#### What were you doing?

Using Filament Manager to keep track of the amount of filament on spools.

#### What did you expect to happen?

When selecting from the list of spools, the list be shared across multiple instances of octoprint, with the database hosted on a server.

#### What happened instead?

Each instance of octoprint hosts it's own database locally.

#### Link to octoprint.log

N/A

#### Link to gcode file

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Further information

Would like the option of hosting the database on a server so that multiple octoprint setups can share the same spools. Maybe having loaded spools be locked by the instance so that collides can't happen.

I have read the FAQ and provided all required information.

malnvenshorn, 2018-02-17:
This is a duplicate of #3. The filament manager already supports PostgreSQL as an alternative to the internal database.

closed 2018-02-17 by malnvenshorn

malnvenshorn, 2018-02-17:
If you need help, there is now a wiki entry on how to setup PostgreSQL.