malnvenshorn/OctoPrint-FilamentManager #85: [Request] Change Dialog to Confirm Filament Selection

bwright86, 2018-03-19:
I have started using the Filament Manager plugin for Octoprint, and it is a very helpful plugin for keeping track of my inventory.

I have enabled the setting to "Show dialog to confirm selected spools before starting/resuming the print job", and noticed that when a print starts, you are required to select the filament that was selected in the left navigation bar. If you do not select the same filament, it does not allow you to proceed with the print.

If the selected filament is incorrect for the print, the user has to:
1. Cancel the dialog
2. Update the filament in the left navigation bar, and
3. Click Print again.

This increases the amount of time it takes to get a print running, if a subsequent prints require different filaments, and the user forgets to update the filament in the left navigation bar.

My suggestion would be to change this to a "Last chance to update your selected filament(s)" dialog, and allow users to see what is currently selected, and if wrong, update the selection to the correct filaments. It would then make the changes, and track the now updated filament selection during the print.

This would remove the need to:
- Cancel the dialog
- Update the filament selection
- Re-click the Print button.

malnvenshorn, 2018-03-19:
This has been discusses several times and I'm not going to change the behavior of the confirmation dialog. The dialog is designed to prevent prints with the wrong spool(s) selected.

You sad yourself "This increases the amount of time it takes to get a print running" this implies that you want to start your print as fast as possible. Now consider the following situation:

* You forgot to change your selected filament in the sidebar
* The dialog opens and you accidentally select the wrong spool because you are in a hurry
* You start the print with the wrong filament

closed 2018-03-19 by malnvenshorn

bwright86, 2018-03-19:
That is fair, if it has already been discussed, I can go along with that.

I was realizing how this worked, and thought of the above to help reduce friction to start a print.
I didn't see previous issues about this feature, so i figured i would write in and share.

Thanks for responding and sharing your viewpoint.