malnvenshorn/OctoPrint-FilamentManager #86: [Request] Power consumption per filament

Graloth, 2018-03-22:
I use the FIlamentManager together with the CostEstimation plugin, which works great.
However, my printer draws different Kw depending on what filament I use, for example, if I use PLA, I run on an overall lower temperature, whereas if I use PETG the temperature is higher and thus the power consumption is also higher.

Therefore, it would be nice, if both plugins are installed, that I could define the Kw power consumption per filament (Or maybe it would fit better on a filament profile).

malnvenshorn, 2018-03-22:
To be honest I consider this to be overkill. Usually the power consumption is the smallest share of the printing costs or does it make a significant difference in your setup? I would just go with the highest/average value.

Graloth, 2018-03-22:
Agree, it would purely be to have accurate data.

malnvenshorn, 2018-04-10:
You will never have 100 percent accurate data. E.g. the plugin has no distinction between preheating and printing, a lot of small moves also increase the power consumption because the stepper motors need more power when moving than just holding their position, etc.

Another point is that I would like to keep both plugins separate, so every user can decide what they want to use. Because of that I'm not comfortible with the idea to add something to the FilamentManager which will be needed by the CostEstimation plugin. 

Therefore I'm closing this issue. But if you have a wattmeter I would be interested in some measurements how different temperatures impact the power consumption of the hotend/print bed.

closed 2018-04-10 by malnvenshorn