malnvenshorn/OctoPrint-FilamentManager #89: Multiple Octoprint Server one Database

veneficusoz, 2018-03-28:
Before opening a new ticket please take a look if there is already an existing one addressing the same concern.

Also read the FAQ: https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki#faq

If you have a **request** or **question** mark them with a "[Request]" respectively "[Question]" prefix in the title.

For **bug** reports use the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the release notes.

(Before submitting your ticket, please delete this text up to and including the line too - it's only here for your information)

---

#### What were you doing?

[Be as specific as possible. Providing a step by step guide will help me to reproduce the issue. This is needed in order to fix the problem. I cannot help you if I don't know what exactly you did to trigger the issue.]

#### What did you expect to happen?

#### What happened instead?

#### Link to octoprint.log

[Always include the log file.]

#### Link to gcode file

[If you encounter issues with the odometer feature provide a small gcode file which produces the problem.]

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

[If you have more information which could help me to understand/fix the issue feel free to add them here.]

I have read the FAQ and provided all required information.

veneficusoz, 2018-03-28:
is it possible to connect 2 Octopi server (running one one Raspberry Pi) connect to one Database ?

malnvenshorn, 2018-03-29:
Yes, see #82

closed 2018-03-29 by malnvenshorn