malnvenshorn/OctoPrint-FilamentManager #91: Filament select befor print dialog box not working

dkmst, 2018-04-03:
I have OctoPi 1.3.7rc3 installed along with filament manager where I have ticked on "Show dialog to confirm selected spools before starting/resuming the print job"

but I am not able or get the option til click "start" is it diable no mater what I do of selection of filament

Menschomat, 2018-04-10:
Same here! Also error's while choosing the spool in sidebar.

Menschomat, 2018-04-10:
2018-04-10 12:02:44,732 - octoprint.plugins.filamentmanager - ERROR - Failed to update selected spool for tool0: (psycopg2.ProgrammingError) FEHLER:  Syntaxfehler bei »ON«
LINE 1: ...ES (0, '7ea3ce62-e05b-11e7-83a9-00fd45fdd2b8', 6) ON CONFLIC...
                                                             ^
 [SQL: 'INSERT INTO selections (tool, client_id, spool_id) VALUES (%(tool)s, %(client_id)s, %(spool_id)s) ON CONFLICT ON CONSTRAINT selections_pkey DO UPDATE SET spool_id = %(param_1)s'] [parameters: {'tool': 0, 'param_1': 6, 'client_id': '7ea3ce62-e05b-11e7-83a9-00fd45fdd2b8', 'spool_id': 6}]
2018-04-10 12:02:44,733 - tornado.access - ERROR - 500 PATCH /plugin/filamentmanager/selections/0 (192.168.0.28) 5.51ms
2018-04-10 12:02:45,686 - tornado.access - WARNING - 409 POST /api/printer/command (192.168.0.28) 3.99ms
2018-04-10 12:04:49,922 - octoprint.plugins.filamentmanager - ERROR - Failed to update selected spool for tool0: (psycopg2.ProgrammingError) FEHLER:  Syntaxfehler bei »ON«
LINE 1: ...ES (0, '7ea3ce62-e05b-11e7-83a9-00fd45fdd2b8', 4) ON CONFLIC...
                                                             ^
 [SQL: 'INSERT INTO selections (tool, client_id, spool_id) VALUES (%(tool)s, %(client_id)s, %(spool_id)s) ON CONFLICT ON CONSTRAINT selections_pkey DO UPDATE SET spool_id = %(param_1)s'] [parameters: {'tool': 0, 'param_1': 4, 'client_id': '7ea3ce62-e05b-11e7-83a9-00fd45fdd2b8', 'spool_id': 4}]
2018-04-10 12:04:49,924 - tornado.access - ERROR - 500 PATCH /plugin/filamentmanager/selections/0 (192.168.0.28) 10.63ms

malnvenshorn, 2018-04-10:
Take a look at #96, seems to be the same issue for me.

Menschomat, 2018-04-10:
Thanks @malnvenshorn  👍 Now working with Postgre 9.6. You should mark the required Version in Readme.md.

Menschomat, 2018-04-10:
@dkmst I hope, this will solve your problem, too? I'm running 1.37-Final

malnvenshorn, 2018-04-10:
> You should mark the required Version in Readme.md.

Already done :wink:

dkmst, 2018-04-10:
@Menschomat i am on 1.3.7rc3 - but will update today...

So what was the solution to the problem - I dont understand half of what there is written in #96

malnvenshorn, 2018-04-10:
@dkmst If you would provide a complete ticket with all requested information I might be able to help.

malnvenshorn, 2018-04-17:
Required information still not provided. Closing.

closed 2018-04-17 by malnvenshorn

dkmst, 2018-04-17:
@Menschomat sry for the delay, i have had a busy time. i will the rest of the things ready to day for you