malnvenshorn/OctoPrint-FilamentManager #92: Plugin installs but fails to show up on plugin configuration menu

garrettbrainard, 2018-04-04:
Before opening a new ticket please take a look if there is already an existing one addressing the same concern.

Also read the FAQ: https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki#faq

If you have a **request** or **question** mark them with a "[Request]" respectively "[Question]" prefix in the title.

For **bug** reports use the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the release notes.

(Before submitting your ticket, please delete this text up to and including the line too - it's only here for your information)

---

#### What were you doing?

Installed FilamentManager

#### What did you expect to happen?

To be able to configure the plugin in the menu

#### What happened instead?

The plugin does not appear in the menu for configuration shown below:
[image]

#### Link to octoprint.log

2018-04-03 03:34:34,248 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.8 2018-04-03 03:39:03,338 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.8 2018-04-03 03:50:57,354 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.6 2018-04-03 04:06:31,161 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.6 2018-04-03 04:06:50,691 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.6 2018-04-03 04:06:50,951 - octoprint.util.comm - INFO - Finished in 4628.658 s. 2018-04-03 04:06:51,087 - octoprint.filemanager.analysis - INFO - Starting analysis of local:MSMV2E3D_C_Dual_RollerPlate.gcode 2018-04-03 04:06:51,181 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/MSMV2E3D_C_Dual_RollerPlate.gcode 2018-04-03 04:12:59,907 - octoprint.filemanager.analysis - INFO - Starting analysis of local:MPCNC_525_Drag_Knife.gcode 2018-04-03 04:12:59,945 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/MPCNC_525_Drag_Knife.gcode 2018-04-03 04:13:24,189 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.6 2018-04-03 04:13:24,838 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.6 2018-04-03 04:13:50,496 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.6 2018-04-03 04:13:51,168 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.6 2018-04-03 04:17:43,447 - octoprint.filemanager.analysis - INFO - Analysis of entry local:MPCNC_525_Drag_Knife.gcode finished, needed 283.54s 2018-04-03 04:17:44,680 - octoprint.filemanager.analysis - INFO - Starting analysis of local:MPCNC_525_Drag_Knife.gcode 2018-04-03 04:17:44,704 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/MPCNC_525_Drag_Knife.gcode 2018-04-03 04:22:11,158 - octoprint.filemanager.analysis - INFO - Analysis of entry local:MPCNC_525_Drag_Knife.gcode finished, needed 266.48s 2018-04-03 04:39:16,911 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/malnvenshorn/OctoPrint-FilamentManager/archive/master.zip 2018-04-03 04:43:43,220 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugin/core.py:1538: UserWarning: No PKG-INFO found for package: octoprint_filamentmanager warnings.warn('No PKG-INFO found for package: %s' % self.package_name) 2018-04-03 04:43:43,286 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager Traceback (most recent call last): File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugin/core.py", line 732, in _import_plugin instance = imp.load_module(key, f, filename, description) File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/__init__.py", line 17, in from .data import FilamentManager File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/data/__init__.py", line 11, in from backports import csv ImportError: cannot import name csv 2018-04-03 04:43:43,474 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint. 2018-04-03 04:50:23,368 - octoprint.server.api.system - INFO - Performing command for core:restart: sudo service octoprint restart 2018-04-03 04:50:23,968 - octoprint.server - INFO - Shutting down... 2018-04-03 04:50:24,750 - octoprint.events - INFO - Processing shutdown event, this will be our last event 2018-04-03 04:50:25,535 - octoprint.events - INFO - Event loop shut down 2018-04-03 04:50:25,569 - octoprint.server - INFO - Goodbye! 2018-04-03 04:50:42,338 - octoprint.startup - INFO - 

There is shown the warning in the code.

[Always include the log file.]

#### Link to gcode file

Not Needed, no print was done

#### Link to contents of Javascript console in the browser
N/A

#### Further information

There is a warning in the log file, I'm running the most current stable release of Octoprint
     OctoPrint 1.3.6 running on OctoPi 0.14.0 

I have read the FAQ and provided all required information.

malnvenshorn, 2018-04-04:
It looks like the requirements for the plugin were not installed correctly. You can try to reinstall the plugin or to install the missing `backports.csv` package manually.

NegativeEnergy, 2018-04-05:
How would I install the backports.csv package

malnvenshorn, 2018-04-05:
You can install the package with 
`<path to the venv>/bin/pip install "backports.csv>=1.0.5,<1.1"`

malnvenshorn, 2018-04-17:
I consider this issue as solved, because I got no further response. Closing.

closed 2018-04-17 by malnvenshorn