malnvenshorn/OctoPrint-FilamentManager #94: Filament Changes (M600) during prints are ignored

MagicManFoli, 2018-04-04:
#### What were you doing?

It seems like swapping filament leads to the first difference being dropped, this is how I reproduced it:

1. Initialise a filament change with M600 (from gcode)
2. Pause octoprint, this enables controls
3. Change filament in filament manager
4. Change "real" filament
5. Continue octoprint
6. Continue printer

#### What did you expect to happen?

I expected both filaments to be reduced in weight according to the printed amount.
Ideally M600 in gcode would be detected and a change dialog would be displayed.

#### What happened instead?

The second filament is decremented, the first one is unmodified.

#### Link to octoprint.log

My installation is at home and can't be accessed, but I'm certain that this is just a missing implementation and not a bug.

#### Link to gcode file

Insert M600 into any random gcode, no additional dependencies needed

#### Link to contents of Javascript console in the browser

...?

#### Further information

Related to #76.

I have read the FAQ and provided all required information.

malnvenshorn, 2018-04-05:
Please provide the log files otherwise your report cannot be processed.

MagicManFoli, 2018-04-05:
MF_Icon_v2.gcode is the gcode that I prepared with a M600 half way through.
If needed I can upload it too, but the described behaviour can be reproduced with any file containing a M600

[octoprint.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1880800/octoprint.txt)

malnvenshorn, 2018-04-06:
I don't see any issue. For the first half 3674 mm filament was used and 1443 mm for the rest.

```
2018-04-02 15:05:46,199 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-02 16:04:02,274 - octoprint.plugins.filamentmanager - INFO - Filament used: 3674.49209 mm (tool0)
2018-04-02 16:29:29,616 - octoprint.util.comm - INFO - Finished in 5023.431 s.
2018-04-02 16:29:32,780 - octoprint.plugins.filamentmanager - INFO - Filament used: 1443.21396 mm (tool0)
```

MagicManFoli, 2018-04-07:
Huh, it seems that I have misinterpreted something in my client. 
The issue can be closed.

closed 2018-04-07 by MagicManFoli