malnvenshorn/OctoPrint-FilamentManager #95: Can't connect to remote Postgres Database

mikepaxton, 2018-04-05:
#### What were you doing?

[Be as specific as possible. Providing a step by step guide will help me to reproduce the issue. This is needed in order to fix the problem. I cannot help you if I don't know what exactly you did to trigger the issue.]

#### What did you expect to happen?
Remotely connect to to Postgresql server.  Postgres database is running on another Pi on local network.
#### What happened instead?
Failed to connect to remote database.  
#### Link to octoprint.log

2018-04-04 22:19:55,520 - octoprint.startup - INFO - ******************************************************************************
2018-04-04 22:19:55,521 - octoprint.startup - INFO - Starting OctoPrint 1.3.6
2018-04-04 22:19:55,522 - octoprint.startup - INFO - ******************************************************************************
2018-04-04 22:19:56,745 - octoprint.startup - INFO - Blacklist processing done
2018-04-04 22:19:56,828 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-04-04 22:19:58,529 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-04-04 22:20:00,130 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2018-04-04 22:20:00,155 - octoprint.plugin.core - INFO - Found 13 plugin(s) providing 13 mixin implementations, 15 hook handlers
2018-04-04 22:20:00,359 - octoprint.server - INFO - Intermediary server started
2018-04-04 22:20:00,360 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-04-04 22:20:00,364 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-04-04 22:20:01,303 - octoprint.plugin.core - INFO - Found 13 plugin(s) providing 13 mixin implementations, 15 hook handlers
2018-04-04 22:20:01,362 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-04-04 22:20:01,363 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-04-04 22:20:01,405 - octoprint.plugins.nautilus - INFO - 0 device(s) will receive notifications...
2018-04-04 22:20:01,406 - octoprint.plugins.nautilus - INFO - Nautilus for OctoPrint, started.
2018-04-04 22:20:01,532 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-04-04 22:20:05,005 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2018-04-04 22:20:08,407 - octoprint.util.pip - INFO - Version of pip is 9.0.3
2018-04-04 22:20:08,415 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: Engine '192.168.1.22' not supported
2018-04-04 22:20:08,421 - octoprint.plugin.core - INFO - Initialized 13 plugin implementation(s)
2018-04-04 22:20:08,434 - py.warnings - WARNING - /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/__init__.py:619: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if value_source != value_target:

2018-04-04 22:20:08,439 - octoprint.plugins.nautilus - INFO - Migrate settings from None to 7.
2018-04-04 22:20:08,680 - octoprint.plugin.core - INFO - 13 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/corewizard
|  Cost Estimation (2.1.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_costestimation
|  CuraEngine (<= 15.04) (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/discovery
|  Filament Manager (0.5.3) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_filamentmanager
|  Nautilus (1.20) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_nautilus
|  Plugin Manager (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/pluginmanager
|  Preheat Button (0.2.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_preheat
|  Print History Plugin (1.2) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_printhistory
|  Printer Stats (1.0.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats
|  Software Update (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/virtual_printer
2018-04-04 22:20:12,036 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1000.0
|    ram: 972271616
|  os:
|    id: linux
|    platform: linux2
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/OctoPrint/venv
2018-04-04 22:20:12,051 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-04-04 22:20:12,078 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2018-04-04 22:20:12,394 - octoprint.server - INFO - Shutting down intermediary server...
2018-04-04 22:20:12,521 - octoprint.server - INFO - Intermediary server shut down
2018-04-04 22:20:12,523 - octoprint.events - INFO - Processing startup event, this is our first event
2018-04-04 22:20:12,524 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2018-04-04 22:20:12,571 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2018-04-04 22:20:12,578 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2018-04-04 22:20:12,598 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on OctoPi4 for SSDP
[... 25 lines cut ...]
#### Link to gcode file
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Further information
Was able to remotely connect to database from Octoprint machine using a simple python psycopg2 connect script. 

#!/usr/bin/python2.4
#
# Small script to show PostgreSQL and Pyscopg together
#

import psycopg2

try:
        conn = psycopg2.connect(database='octoprint_filamentmanager', user='octoprint', password='mypasswd', host='192.168.1.22', port='5432')
except:
    print "I am unable to connect to the database"

Looking at the Filament manager connection script it appears to be using SqlAlchemy and not psycopg2 as documented.

I have read the FAQ and provided all required information.
Yes

malnvenshorn, 2018-04-05:
Your database URI is wrong. You should take a look in the [wiki](https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki/Setup-PostgreSQL-on-Arch-Linux#plugin-settings).

mikepaxton, 2018-04-05:
Sorry, I forgot to change that back before restarting octoprint and sending the log.  I was trying various combinations of the URI to get it to work.  
Even when using the correct URI I'm still having the issues.  Here is a fresh log with the corrrect URI. 

2018-04-05 10:47:38,986 - octoprint.startup - INFO - ******************************************************************************
2018-04-05 10:47:38,987 - octoprint.startup - INFO - Starting OctoPrint 1.3.6
2018-04-05 10:47:38,988 - octoprint.startup - INFO - ******************************************************************************
2018-04-05 10:47:40,000 - octoprint.startup - INFO - Blacklist processing done
2018-04-05 10:47:40,056 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-04-05 10:47:41,830 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-04-05 10:47:43,480 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2018-04-05 10:47:43,505 - octoprint.plugin.core - INFO - Found 13 plugin(s) providing 13 mixin implementations, 15 hook handlers
2018-04-05 10:47:43,716 - octoprint.server - INFO - Intermediary server started
2018-04-05 10:47:43,717 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-04-05 10:47:43,721 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-04-05 10:47:44,694 - octoprint.plugin.core - INFO - Found 13 plugin(s) providing 13 mixin implementations, 15 hook handlers
2018-04-05 10:47:44,755 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-04-05 10:47:44,757 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-04-05 10:47:44,797 - octoprint.plugins.nautilus - INFO - 0 device(s) will receive notifications...
2018-04-05 10:47:44,798 - octoprint.plugins.nautilus - INFO - Nautilus for OctoPrint, started.
2018-04-05 10:47:45,013 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-04-05 10:47:48,444 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2018-04-05 10:47:51,877 - octoprint.util.pip - INFO - Version of pip is 9.0.3
2018-04-05 10:47:51,897 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: No module named psycopg2
2018-04-05 10:47:51,903 - octoprint.plugin.core - INFO - Initialized 13 plugin implementation(s)
2018-04-05 10:47:51,917 - py.warnings - WARNING - /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/__init__.py:619: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if value_source != value_target:

2018-04-05 10:47:51,929 - octoprint.plugin.core - INFO - 13 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/corewizard
|  Cost Estimation (2.1.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_costestimation
|  CuraEngine (<= 15.04) (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/discovery
|  Filament Manager (0.5.3) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_filamentmanager
|  Nautilus (1.20) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_nautilus
|  Plugin Manager (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/pluginmanager
|  Preheat Button (0.2.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_preheat
|  Print History Plugin (1.2) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_printhistory
|  Printer Stats (1.0.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats
|  Software Update (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/virtual_printer
2018-04-05 10:47:55,337 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1000.0
|    ram: 972271616
|  os:
|    id: linux
|    platform: linux2
|  python:
|    pip: 9.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/OctoPrint/venv
2018-04-05 10:47:55,352 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-04-05 10:47:55,381 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
[... cut ...]
2018-04-05 10:48:46,072 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.124
2018-04-05 10:48:50,379 - octoprint.plugins.filamentmanager - ERROR - Failed to fetch profiles lastmodified timestamp: 'NoneType' object has no attribute 'get_profiles_lastmodified'
2018-04-05 10:48:50,380 - octoprint.plugins.filamentmanager - ERROR - Failed to fetch profiles: 'NoneType' object has no attribute 'get_all_profiles'
2018-04-05 10:48:50,387 - tornado.access - ERROR - 500 GET /plugin/filamentmanager/profiles (127.0.0.1) 16.91ms

malnvenshorn, 2018-04-17:
I consider this issue as solved, because I got no further response. Closing.

closed 2018-04-17 by malnvenshorn