malnvenshorn/OctoPrint-FilamentManager #97: filament length and weight not cleared when no print sellected

TomW1605, 2018-04-12:
#### What were you doing?
viewing the server with no file loaded
#### What did you expect to happen?
i expected to see "Filament: - / -"
#### What happened instead?
"Filament: 1.86m / 5.19g"
#### Link to octoprint.log
[octoprint (4).log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/1903233/octoprint.4.log)
#### Link to gcode file
#### Link to contents of Javascript console in the browser

This page includes a password or credit card input in a non-secure context. A warning has been added to the URL bar. For more information, see https://goo.gl/zmWq3m.
packed_core.js?ccbe5583:14200 Starting dependency resolution...
packed_plugins.js?ba1de181:17236 THREE.WebGLRenderer 90
packed_core.js?ccbe5583:14310 ... dependency resolution done
packed_core.js?ccbe5583:14613 Initial application setup done, connecting to server...
packed_core.js?ccbe5583:12559 Connected to the server
packed_core.js?ccbe5583:14591 Finalizing application startup
5packed_plugins.js?ba1de181:19103 THREE.MultiMaterial has been removed. Use an Array instead.
MultiMaterial @ packed_plugins.js?ba1de181:19103
packed_core.js?ccbe5583:14475 Going to bind 39 view models...
packed_core.js?ccbe5583:14528 Did not bind view model CuraViewModel to target #wizard_plugin_cura since it does not exist
packed_core.js?ccbe5583:14528 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?ccbe5583:2300 User Tom logged in
packed_core.js?ccbe5583:14567 ... binding done
packed_core.js?ccbe5583:14587 Application startup complete
packed_plugins.js?ba1de181:19103 THREE.MultiMaterial has been removed. Use an Array instead.

#### Further information
this is also a problem when switching to an unanalysed file. it will keep showing the values for the last file till the new one is analysed.

I have read the FAQ and provided all required information.

malnvenshorn, 2018-04-14:
Fixed by the above commit. Available on `develop` branch and will be released with 0.6.0

TomW1605, 2018-04-14:
@malnvenshorn thanks. how long till 0.6.0 is released? just wondering if it is worth updating or should i just wait?

commit 3813865c (2018-04-14): Filament info is reset if current file is null (Fixes #97)