malnvenshorn/OctoPrint-FilamentManager #98: [Request] add a spool reset button that can be used mid print

TomW1605, 2018-04-14:
i think this would be useful for when you are in the middle of a print and need to switch to a new spool with the same properties as the current one because it has run out. it should be relatively simple to implement.to be clear i am not asking for any functionality relating to M600 commands simply a button that resets the current spool to it's starting weight.

TomW1605, 2018-04-14:
i think this is what was being requested in issue #83 but i'm not sure

malnvenshorn, 2018-04-15:
You can simply duplicate the spool (which also resets the used filament on it) and select the new one.

TomW1605, 2018-04-15:
the problem with that is you can't do it mid print without pausing.

malnvenshorn, 2018-04-15:
Yes that's true, but to be able to change the filament you have to pause the print anyway.

TomW1605, 2018-04-15:
only if you are changing to a different filament and need to clean the nozzle. if you are switch to a new spool of the same filament because the old one has run out then you don't.

malnvenshorn, 2018-04-15:
Yes you are right, but it also can go wrong. It's safer the pause the print. But if you don't want that why don't you select a spool with sufficient filament for the print?

TomW1605, 2018-04-15:
i usually only run 1 spool at a time

TomW1605, 2018-04-15:
if i can work out a way to do it and submit a pull request will you look at it?

TomW1605, 2018-04-16:
ok you are probably right about other people not using it. i am vary new to 3d printing and am mostly making it up as i go so i tend to do things like hotswaping filament that most people have decided is a bad idea. i will make a new fork and see what i come up with but given i dont swap filament that often i doubt i will put much efort into it