malnvenshorn/OctoPrint-FilamentManager #103: [Question] Any way to combine with PrintHistory plugin?

Wolfsblvt, 2018-05-08:
Hey.
Really nice plugin you have here! I am very likely going to use it.

Currently, I use the [PrintHistory](https://github.com/imrahil/OctoPrint-PrintHistory) plugin by imrahil to track filament usage and my spools manually.

Any chance to combine those two plugins, getting a cool history and cool filament tracker?

malnvenshorn, 2018-05-09:
Since the mentioned plugin is an abandoned project it's unlikely that they will work together some day. But the good news is that there is already another history plugin in the pipeline, see https://discourse.octoprint.org/t/wip-new-print-history-plugin/1633 for more information :wink:

Wolfsblvt, 2018-05-09:
Uh. That sounds even better. I have to follow this project then.
Thank you!

closed 2019-02-28 by malnvenshorn